flipperdevices/flipperzero-firmware #2278: fbt documentation: pip "Requirements" should be removed/clarified

Fasjeit, 2023-01-08:
### Describe the bug.

As mentioned in #2253 and #2060 running 'fbt' is more than enough to install al requirements, so the Requirements section in [fbt.md](https://github.com/flipperdevices/flipperzero-firmware/blob/dev/documentation/fbt.md#requirements) is misleading and no longer needed or should be clarified.

Trying to run `pip3 install -r scripts/requirements.txt` on windows 10 cause the following error

```
Microsoft Visual C++ 14.0 or greater is required. Get it with "Microsoft C++ Build Tools":
https://visualstudio.microsoft.com/visual-cpp-build-tools/
```

after installing ~7 GB of Windows SDK followed by 
```
Running setup.py install for heatshrink2 ... error
  error: subprocess-exited-with-error

  × Running setup.py install for heatshrink2 did not run successfully.
  │ exit code: 1
  ╰─> [88 lines of output]
...
core.c
      heatshrink2/core.c(5406): warning C4018: '<': signed/unsigned mismatch
      heatshrink2/core.c(10933): error C2037: left of 'f_lineno' specifies undefined struct/union '_frame'
      heatshrink2/core.c(13136): error C2039: 'exc_type': is not a member of '_err_stackitem'
...
× Encountered error while trying to install package.
╰─> heatshrink2
```

And it turns out trying to run this command was useless (after total 6 hours of trying) and I can just run

```
./fbt.cmd
```

### Reproduction

Run `pip3 install -r scripts/requirements.txt` on windows 10.

### Logs

```Text
PS C:\git\unleashed-firmware> pip install -r scripts/requirements.txt
Collecting ansi==0.3.6
  Using cached ansi-0.3.6-py3-none-any.whl (8.6 kB)
Collecting black==22.6.0
  Using cached black-22.6.0-py3-none-any.whl (156 kB)
Collecting colorlog==6.7.0
  Using cached colorlog-6.7.0-py2.py3-none-any.whl (11 kB)
Collecting heatshrink2==0.11.0
  Using cached heatshrink2-0.11.0.tar.gz (110 kB)
  Preparing metadata (setup.py) ... done
Collecting Pillow==9.1.1
  Using cached Pillow-9.1.1.tar.gz (49.8 MB)
  Preparing metadata (setup.py) ... done
Collecting protobuf==3.20.1
  Using cached protobuf-3.20.1-py2.py3-none-any.whl (162 kB)
Requirement already satisfied: pyserial==3.5 in c:\users\fasjeit\appdata\local\programs\python\python311\lib\site-packages (from -r scripts/requirements.txt (line 7)) (3.5)
Collecting python3-protobuf==2.5.0
  Using cached python3-protobuf-2.5.0.tar.gz (69 kB)
  Preparing metadata (setup.py) ... done
Collecting SCons==4.4.0
  Using cached SCons-4.4.0-py3-none-any.whl (4.2 MB)
Collecting click>=8.0.0
  Using cached click-8.1.3-py3-none-any.whl (96 kB)
Collecting platformdirs>=2
  Using cached platformdirs-2.6.2-py3-none-any.whl (14 kB)
Collecting pathspec>=0.9.0
  Using cached pathspec-0.10.3-py3-none-any.whl (29 kB)
Requirement already satisfied: mypy-extensions>=0.4.3 in c:\users\fasjeit\appdata\local\programs\python\python311\lib\site-packages (from black==22.6.0->-r scripts/requirements.txt (line 2)) (0.4.3)
[... 85 lines cut ...]
      heatshrink2/core.c(13542): error C2039: 'exc_traceback': is not a member of '_err_stackitem'
      C:\Users\Fasjeit\AppData\Local\Programs\Python\Python311\include\cpython/pystate.h(55): note: see declaration of '_err_stackitem'
      heatshrink2/core.c(13548): error C2037: left of 'f_back' specifies undefined struct/union '_frame'
      heatshrink2/core.c(13854): error C2039: 'exc_type': is not a member of '_err_stackitem'
      C:\Users\Fasjeit\AppData\Local\Programs\Python\Python311\include\cpython/pystate.h(55): note: see declaration of '_err_stackitem'
      heatshrink2/core.c(13854): error C2198: 'visit': too few arguments for call
      heatshrink2/core.c(13856): error C2039: 'exc_traceback': is not a member of '_err_stackitem'
      C:\Users\Fasjeit\AppData\Local\Programs\Python\Python311\include\cpython/pystate.h(55): note: see declaration of '_err_stackitem'
      heatshrink2/core.c(13856): error C2198: 'visit': too few arguments for call
      heatshrink2/core.c(14103): error C2039: 'exc_type': is not a member of '_err_stackitem'
      C:\Users\Fasjeit\AppData\Local\Programs\Python\Python311\include\cpython/pystate.h(55): note: see declaration of '_err_stackitem'
      heatshrink2/core.c(14105): error C2039: 'exc_traceback': is not a member of '_err_stackitem'
      C:\Users\Fasjeit\AppData\Local\Programs\Python\Python311\include\cpython/pystate.h(55): note: see declaration of '_err_stackitem'
      error: command 'C:\\Program Files (x86)\\Microsoft Visual Studio\\2022\\BuildTools\\VC\\Tools\\MSVC\\14.34.31933\\bin\\HostX86\\x64\\cl.exe' failed with exit code 2
      [end of output]

  note: This error originates from a subprocess, and is likely not a problem with pip.
error: legacy-install-failure

× Encountered error while trying to install package.
╰─> heatshrink2

note: This is an issue with the package mentioned above, not pip.
hint: See above for output from the failure.
```

peoblouk, 2023-01-21:
I have same problem, how did you solve it??

Fasjeit, 2023-01-22:
Just ignore that line and run
```
./fbt
```

in project directory in bash or powershell. This will install the toolchain with all required dependencies.

If you are planning to use vs code you will also need to

1. run
```
./fbt vscode_dist
./fbt firmware_cdb
```

2. Install all suggested extensions for the project (vs code will ask you to do it when you open the project folder).
3. If you are using Windows with backmagic probe you need to adjust `launch.json` file like this (https://github.com/flipperdevices/flipperzero-firmware/pull/2282), To make gdb silent during blackmagic flashing you probably want to edit `fbt_debugopts.py` - add this line at the end of `GDBOPTS_BLACKMAGIC` : 
```
"--batch",
```

skotopes, 2023-02-10:
We are planning to drop requirements file from firmware repository and move it to toolchain folder. Probably in next update.

skotopes, 2023-03-13:
Removed in #2420

closed 2023-03-13 by skotopes