flipperdevices/flipperzero-firmware #2279: Cannot parse key file created by flipper himself

Salamafet, 2023-01-09:
### Describe the bug.

According to this [thread](https://forum.flipperzero.one/t/cannot-parse-file-created-from-the-flipper-sub-ghz-read-menu/11317), I can parse a sub-ghz key file create with the Flipper himself.

Here two example button of the same remote:
```
Filetype: Flipper SubGhz Key File
Version: 1
Frequency: 433920000
Preset: FuriHalSubGhzPresetOok650Async
Protocol: CAME
Bit: 42
Key: 00 00 03 77 BE E9 8D 0B
```
```
Filetype: Flipper SubGhz Key File
Version: 1
Frequency: 433920000
Preset: FuriHalSubGhzPresetOok650Async
Protocol: CAME
Bit: 42
Key: 00 00 03 77 BF 68 2C 13
```

### Reproduction

1. Read the key from the Sub-ghz menu
2. Save the key
3. Replay the saved key

Get the "Cannot parse file" error
[image]

Skorpionm, 2023-01-09:
when did you create this file?  exactly this file was created on the firmware?  for the CAME protocol, I know 12 bits 23 24 25. It cannot be 42 bits or it is some other protocol.

Salamafet, 2023-01-12:
Create with the firmware 0.74.2 and got the error with 0.74.2 and 0.75.

Here is 4 photos of the remote
[photos_remote.pdf](https://github.com/flipperdevices/flipperzero-firmware/files/10400668/remote.pdf)

Screenshots of the Flipper screen during the normal Read
[image]
[image]
[image]

Raw files:

Button 1:
```
Filetype: Flipper SubGhz RAW File
Version: 1
Frequency: 433657070
Preset: FuriHalSubGhzPresetOok650Async
Protocol: RAW
RAW_Data: 9849 -570 165 -300 233 -132 131 -1000 131 -4304 65 -64 231 -300 165 -66 393 -68 463 -100 863 -100 1065 -66 331 -100 665 -134 369 -66 337 -100 14517 -136 265 -166 163 -202 959 -66 627 -66 335 -68 669 -166 437 -66 3393 -68 331 -68 297 -66 8311 -236 237 -134 65 -134 231 -132 65 -164 99 -66 231 -66 131 -826 67 -132 65 -3618 67 -132 97 -162 97 -390 459 -100 333 -100 199 -136 167 -100 197 -68 1593 -68 695 -66 403 -168 661 -100 197 -136 333 -960 65 -68 163 -1218 97 -1160 233 -166 133 -98 229 -98 525 -132 231 -100 199 -932 229 -166 65 -136 469 -102 133 -66 233 -68 769 -66 1583 -164 599 -100 299 -68 7593 -100 535 -698 297 -98 199 -134 65 -66 65 -200 97 -264 99 -266 165 -134 201 -164 133 -430 65 -690 97 -168 201 -266 67 -66 661 -1490 1483 -66 1411 -238 133 -132 199 -864 165 -68 65 -66 165 -200 65 -136 229 -3060 65 -98 65 -1988 99 -132 131 -268 235 -430 231 -134 65 -134 167 -168 267 -100 297 -166 399 -132 97 -100 401 -200 165 -332 265 -98 235 -134 497 -166 1933 -68 7287 -398 129 -164 133 -198 195 -228 197 -396 131 -6618 165 -100 233 -134 265 -134 267 -134 827 -98 435 -98 167 -66 269 -100 1193 -100 425 -100 327 -98 231 -68 2111 -130 7307 -430 65 -424 65 -132 65 -1024 99 -3888 197 -194 99 -66 231 -166 333 -66 999 -132 99 -68 99 -100 197 -98 327 -98 163 -66 65 -66 199 -100 331 -68 633 -166 395 -100 7311 -266 163 -368 165 -202 433 -66 165 -200 499 -262 97 -4652 165 -198 131 -264 163 -98 883 -98 2115 -100 631 -100 1391 -100 1251 -66 6713 -366 263 -1060 231 -134 365 -166 133 -6238 197 -464 65 -98 261 -162 65 -98 163 -98 195 -494 625 -200 65 -100 497 -164 529 -132 65 -130 817 -66 259 -64 691 -136 697 -98 7771 -398 263 -66 399 -338 99 -596 133 -166 267 -332 327 -66 661 -100 265 -136 265 -134 301 -1132 231 -98 235 -400 197 -200 165 -200 231 -202 99 -232 329 -100 231 -168 1129 -7302 65 -132 65 -894 97 -566 65 -66 199 -562 263 -198 129 -266 297 -266 133 -132 67 -132 67 -166 165 -66 363 -66 97 -330 333 -100 395 -98 1027 -102 265 -66 1851 -98 947 -66 5577 -4030 165 -166 263 -132 325 -132 131 -164 263 -760 493 -164 519 -98 1353 -4094 267 -300 197 -430 65 -134 133 -132 265 -102 301 -166 65 -134 1229 -132 131 -132 267 -232 765 -68 463 -66 299 -132 333 -98 263 -68 363 -100 267 -100 727 -68 11923 -66 97 -396 99 -100
RAW_Data: 233 -202 433 -528 427 -98 261 -296 797 -198 4143 -66 12747 -392 97 -424 163 -594 295 -200 99 -432 199 -562 65 -166 101 -334 165 -200 497 -368 559 -300 197 -300 395 -68 265 -66 3587 -68 997 -66 631 -298 129 -98 231 -98 65 -196 393 -264 131 -66 165 -296 131 -198 63 -558 65 -2878 197 -230 63 -362 99 -166 297 -132 197 -100 1065 -100 165 -134 901 -66 15347 -66 63 -394 465 -66 199 -428 99 -98 393 -396 17411 -102 197 -560 295 -858 195 -230 97 -962 527 -100 865 -232 659 -328 199 -166 199 -366 65 -98 4055 -134 299 -130 197 -68 301 -134 235 -102 1023 -66 8433 -192 131 -6372 197 -98 129 -130 65 -98 197 -66 131 -98 99 -296 295 -64 195 -1182 329 -526 163 -594 65 -1054 329 -66 735 -98 7275 -166 165 -960 197 -236 99 -430 231 -132 165 -5488 65 -1252 131 -230 163 -232 227 -162 331 -198 65 -100 333 -168 195 -1196 233 -202 397 -66 133 -166 1397 -100 6865 -2810 65 -2028 131 -858 165 -300 135 -200 167 -132 133 -198 233 -166 695 -134 235 -166 925 -100 5051 -100 491 -64 195 -66 425 -230 915 -564 99 -66 167 -1064 165 -134 233 -1148 265 -434 297 -264 361 -232 333 -134 397 -132 263 -100 195 -98 163 -100 163 -132 229 -164 197 -98 229 -132 229 -202 1267 -100 633 -100 6097 -198 131 -164 329 -66 327 -1706 495 -134 231 -134 465 -198 165 -134 299 -134 333 -930 165 -398 131 -164 231 -132 995 -100 267 -132 367 -134 1363 -68 8719 -66 131 -396 99 -662 197 -526 99 -1478 263 -132 231 -820 165 -492 229 -332 229 -1256 165 -166 133 -100 67 -134 131 -166 433 -200 401 -132 231 -166 12383 -764 99 -566 199 -198 99 -132 233 -66 533 -300 99 -400 893 -330 165 -66 131 -166 201 -166 295 -96 591 -364 697 -100 99 -100 559 -66 261 -264 295 -264 295 -100 495 -134 201 -266 12297 -794 65 -166 133 -1518 167 -702 65 -100 229 -202 429 -232 265 -66 265 -132 331 -98 199 -102 301 -400 99 -598 197 -98 101 -366 301 -332 493 -100 361 -132 265 -164 97 -66 8139 -4070 259 -730 365 -134 331 -66 267 -134 265 -234 297 -98 199 -498 397 -460 391 -100 97 -66 65 -460 231 -234 397 -132 425 -66 749 -164 457 -66 10969 -786 65 -98 131 -526 97 -98 163 -296 65 -688 265 -98 395 -100 197 -170 565 -166 229 -134 99 -100 97 -66 233 -198 331 -1064 67 -198 901 -308 657 -314 649 -294 669 -316 661 -300 649 -302 661 -320 661 -300 647 -300 659 -320 663 -300 649 -310 655 -314 659 -294 665 -318 629 -5150 641 -290
RAW_Data: 685 -314 625 -340 651 -296 663 -316 629 -304 685 -296 663 -322 631 -292 669 -312 655 -312 651 -294 665 -312 665 -320 645 -290 643 -5120 677 -320 643 -288 681 -314 625 -340 651 -312 623 -316 663 -294 667 -312 661 -312 649 -294 667 -310 665 -282 671 -290 669 -304 659 -320 631 -4818 315 -638 317 -622 345 -280 685 -628 321 -640 303 -630 351 -278 663 -644 315 -636 351 -600 329 -650 311 -306 665 -622 323 -644 309 -634 351 -626 323 -642 317 -320 625 -640 353 -622 331 -622 345 -268 669 -622 323 -316 649 -310 657 -312 649 -294 665 -624 323 -316 643 -3 [... cut ...]  97 -1456 233 -66 65 -1360 263 -100 261 -298 165 -1030 65 -492 163 -98 65 -2478 133 -268 65 -134 565 -68 165 -366 329 -228 231 -232 7081 -4160 165 -132 65 -2406 165 -1324 265 -98 199 -132 235 -1198 165 -166 233 -132 167 -458 229 -1484 229 -800 425 -164 433 -68 1593 -132 7055 -4980 65 -594 97 -66 163 -1518 263 -130 65 -956 263 -296 231 -98 1153 -166 65 -366 5449 -2298 97 -6498 65 -1452 165 -230 327 -132 259 -66 65 -98 99 -558 165 -162 65 -656 65 -430 299 -98 199 -100 331 -66 333 -100 357 -194
```

Skorpionm, 2023-01-12:
I repeat once again you do not have a stock version of the firmware, put the stock one and record 5 presses of each button and holding for 3 seconds

Eugst, 2023-02-11:
Facing the same trouble. Firmware from the box. Tried 5 times per 3 second - didn't help.

skotopes, 2023-03-03:
Issue was fixed in #2443 please check and let us know if it still persist

closed 2023-03-03 by skotopes