flipperdevices/flipperzero-firmware #2294: Bluetooth BadUSB

yodaluca23, 2023-01-11:
### Describe the enhancement you're suggesting.

Let us run normal BadUSB scripts, over Bluetooth, it should definitely be possible, since we already have BadUSB and a Bluetooth controller keyboard. It just needs to be combined! BadBluetooth?

Faddy96, 2023-01-11:
Why do you want this? I don’t really see the point of it, not even as a convenience tool for testing BadUSB scripts on your own PC.

yodaluca23, 2023-01-11:
> Why do you want this? I don’t really see the point of it, not even as a convenience tool for testing BadUSB scripts on your own PC.
> 
> 
> 
>  

It would be just another way of doing it, and that is, kinda the whole point of the flipper... Plus then you could use BadUSB scripts without a wire...

Faddy96, 2023-01-12:
The entire premise of a "BadUSB" is that utilizes the bad security of USB-ports on computers in which any USB device can tell the computer that it is an USB keyboard. This will allow the attacker to quickly and covertly run payloads (scripts) on their victims, because you can plug in the BadUSB when nobody is watching, wait 5-60 seconds and plug it out. Quick plug-in, quick plug-out.

Bluetooth is a technology that requires both devices to consent to pairing. So if you want to carry out a BadUSB script over Bluetooth, you will have to sit down and use the victim's computer to set up the pairing. And if you want to run BadUSB scripts without a wire on your own PC, I mean you're gonna sit in front of the PC anyways so just plug it in. And if you want to test your scripts and modify them, you will still need to plug the Flipper with wire so that you can download, edit, and re-upload the file to your flipper.

The only motivation I see for this is that you forgot to bring a cable with you. Or you want to prank someone when they're gone so you spend 5 minutes to pair your Flipper and then wait for the victim to come back so that you can Rick Roll them while they are focused on doing something.

Alsa2, 2023-01-16:
Maybe you could change the name to impersonate an object needing Bluetooth (speakers, headphone…) and execute the keystrokes once paired.

kowalski7cc, 2023-01-30:
I think the only thing that would make sense for something like that would be some macros in the stream deck style, but it would require some desktop side integration. Maybe with qFlipper?

skotopes, 2023-02-10:
We may come back to this idea in firmware version 2.0. Meanwhile this feature can be implemented in external application

closed 2023-02-10 by skotopes

yodaluca23, 2023-09-11:
It has been implemented into a FAP and added to RM Firmware.
https://github.com/RogueMaster/flipperzero-firmware-wPlugins/pull/620