flipperdevices/flipperzero-firmware #2295: [NFC] Magic Tag emulation

ComicBit, 2023-01-11:
### Description of the feature you're suggesting.

It would be very useful to have the possibility of mimic a magic tag behaviour so that the emulation could work in every enviroment

skotopes, 2023-02-10:
Are we talking about MFC, and you mean allow write to sector 0?

ComicBit, 2023-02-10:
> Are we talking about MFC, and you mean allow write to sector 0?

Yes

skotopes, 2023-02-10:
That is possible and we will accept PR if someone implement it.

skotopes, 2023-02-10:
Meanwhile I'll put this task to backlog.

skotopes, 2024-07-07:
After some thought we decided to not to implement it. Too much troubles for very niche things.

closed 2024-07-07 by skotopes

ComicBit, 2024-07-08:
This is very sad :( that would have been one of the best functionalities ever for testing

skotopes, 2024-07-08:
@ComicBit maybe someone from community will do it, maybe it will be you ;-)

ComicBit, 2024-07-08:
> @ComicBit maybe someone from community will do it, maybe it will be you ;-)

I'd love to! But I have not much experience with programming such things