flipperdevices/flipperzero-firmware #2297: Architecture support

detectingdragon77, 2023-01-13:
### Description of the feature you're suggesting.

Arm64 linux toolchain support would be an amazing addition.

FalsePhilosopher, 2023-01-17:
From what I remember they ported fbt to arm64 and you just run it like on an x86 machine.

detectingdragon77, 2023-01-17:
> From what I remember they ported fbt to arm64 and you just run it like on an x86 machine.

There is no toolchain support for arm64.

skotopes, 2023-02-10:
It should be possible to compile toolchain for arm linux, but supporting one more architecture is quite painful for us.
We are not planning to do it until critical mass is reached.

Meanwhile fbt supports 3rd party toolchains, so in theory you can build your own or just install components into the system.

closed 2023-02-10 by skotopes