flipperdevices/flipperzero-firmware #2298: [NFC] Reading some types of tag stops emulation from working until certain other types are read

TollyH, 2023-01-14:
### Describe the bug.

After reading an EMV or DESFire NFC card, devices will no longer detect the Flipper when emulating other types of NFC (i.e. NTAG). Scanning a Mifare Classic or an NTAG, or rebooting the Flipper, will fix the issue allowing emulation to be detected again.

### Reproduction

1. Scan either an EMV card or a DESFire card as normal from the NFC menu
2. Now attempt to emulate an NTAG device (like an Amiibo) from either the NFC menu, or the Archive
3. The Flipper will appear to emulate as normal without any log errors, but devices will fail to detect it
4. Either reboot the Flipper, or scan either an NTAG device or Mifare Classic card
5. Once again start emulating an NTAG device
6. Devices will now detect the Flipper as normal again

skotopes, 2023-02-10:
Yep this is known issue. There were partial fix for it in previous releases and it will be fully fixed after NFC HAL refactoring.
Please let us know if issue persist for you in 0.77.

TollyH, 2023-02-10:
At the moment I'm still having this specific issue in 0.77.0-rc

skotopes, 2023-02-10:
@TollyH thanks, we'll try to address it next update

nvx, 2023-02-13:
Is there a planned timeline for the NFC HAL refactoring @skotopes, or is it just on the todo list?

skotopes, 2023-02-13:
Yes, it's currently work in progress.
As for specific time line: we are planning to open base part of new driver with MFC in one month. Then other protocols.

skotopes, 2023-03-13:
Should be fixed in previous release. Please reopen if you still have this issue.

closed 2023-03-13 by skotopes