flipperdevices/flipperzero-firmware #2301: Manual data I/O for OneWire with iButtonWorker

team-orangeBlue, 2023-01-14:
### Description of the feature you're suggesting.

Flipper's iButton implementation uses iButtonWorker, which appears to be very vague due to currently lacking documentation (example: `/applications/main/ibutton/scene/ibutton_scene_write.c` at line 66 with command `ibutton_worker_write_start` - What the heck is the write command??? 0x1D or something else?).

Due to these limitations, support for memory iButtons like DS1996 (code 0x0C, 8KB iButton) which are commonly used in Russia for data transfer or TM01 (universal blank without high voltage requirements) becomes impossible.
A command to manually send or receive data over OneWire with iButtonWorker would be highly appreciated.

### Anything else?

Datasheets for DS1996 and TM01 are linked below:
https://cdn.discordapp.com/attachments/954430078882816021/1063194542469300304/ds1996.pdf
https://cdn.discordapp.com/attachments/954430078882816021/1063194542133743677/TM-01C.doc
**Caution!** This TM01 datasheet does NOT contain:
• Irreversible lock bit configuration
• TM01 configuration (voltage level, delay)

skotopes, 2023-02-10:
ibutton is currently being refactored, we'll return to this issue after refactoring complete

team-orangeBlue, 2023-02-10:
> ibutton is currently being refactored, we'll return to this issue after refactoring complete

Do you think adding support for RW2017 or similar would be possible? I can try to make a protocol datasheet with my TMD

skotopes, 2023-02-10:
That is goal for this refactoring.

team-orangeBlue, 2023-03-07:
Alright, I've flashed last dev firmware with this patch pushed. Thanks for DS1996, really nice improvement! I should get my RW2017 analyzed this thursday.
Emulation however appears to have taken a hit and does not show any visible improvements. Maybe it's because of my crappy "real ibutton" mod which shorted 1W to GND, I have though tried a couple without it with fails. I will redo checks on a completely stock, dev fw flipper.

About that hit.. the entirety of flipper's telegram knows that the flipper cannot function with Vizit and Eltis readers. Marshal+/ДомофонСБ/XVoice is also there. PR #2388 should have fixed some emulation filters, but from my today's tests it seems to have actually made it worse.
Below is a video of a Cyfral intel. (it just works.)
https://user-images.githubusercontent.com/63470411/223556281-7da65422-d969-4097-9f35-0c58695d26b1.mp4
And now post-update. No reactions at all. If the code was removed, a 4-part buzzer would sound
https://user-images.githubusercontent.com/63470411/223556287-da45aa0e-a875-4e3e-b2e3-076709ccef40.mp4

What hasn't changed (at least I think so..): Vizit does not appear to work. Tested on SM100, SM101, BVD-431DXKCB. 
Uploading VID_20230308002650.mp4…
Eltis DP does not work. Or does it? The code I tapped was enrolled. _Damn it! Should have added my keys when it was probably on autowrite._
https://user-images.githubusercontent.com/63470411/223557216-58ae116b-1dc0-485e-bed7-832864cfb2b6.mp4
And that was the only case when emulation worked. I tried another DP but it didn't work.

Below is a table of what changed after this update as far as I can tell
| **DEVICE** | **FUNCTIONALITY** | **VERDICT** |
| --- | --- | --- |
| Cyfral v2 | Untested, probably works | / |
| Cyfral v3 | Untested, probably works | / |
| Cyfral "StroyMir" | No longer works on Cyfral Intel | X |
| Metakom v1 | None near me, can't test, probably works | / |
| Metakom v2 | Works, both DS1990 and TM2002 | ✓ |
| Vizit (any) | Does not work at all | X |
| Eltis CP100 | Appears to have stopped working; will recheck | /X |
| Eltis DPxxx | Functionality poor | X/ |
| Eltis controller | Still not working | X |
| Marshal CD-2255 | Still flashing / symbol | X |
| Marshal CD-7000 | Still works | ✓ |
| KCdomofon 2006/2016d | Can't check, none nearby, 2006 was proven functional before | / |
| Olevs | Unchecked, was functional | ✓/ |
| Raikmann | Can't check, none nearby | / |
| FERMAX | has anyone seen a fermax with TM? | / |
| Laskomex EC2502 | Can't check, none nearby | / |
| UDGin | God knows, I'm in Moscow and not SPB, also can't check nor find! | / |
| Z-5R | Unchecked, probably still works | ✓/ |

✓ means works, / means unsure, X means not functional. If 2 characters are in the verdict (example: AB), then verdict A is considered to be mostly correct, but verdict B has traces.

gsurkov, 2023-03-07:
@team-orangeBlue 

>PR https://github.com/flipperdevices/flipperzero-firmware/pull/2388 should have fixed some emulation filters

No, it should have not. Its sole purpose was to simplify adding more iButton key types, which was nearly impossible in the previous implementation.

If there are regressions (i.e. stuff worked before, doesn't work now), please describe them in a separate issue, try to keep it simple and clear for best results. Statements like "Probably works" do not help at all, so just omit the entries you aren't sure about.

Readers that didn't work in the previous version should go into yet another issue. (No guarantee that this will be ever worked on, but it would be nice to have a wishlist).

We will be also performing our own testing before the release, which will ensure that it works just as well at least on our selection of readers.

gsurkov, 2023-03-09:
@team-orangeBlue 
By the way, does #2388 fix the original cause of this issue? Can we close it?

team-orangeBlue, 2023-03-09:
> @team-orangeBlue By the way, does #2388 fix the original cause of this issue? Can we close it?

Probably. I'll close this for now, maybe gonna open a pull request whenever I get info on the RW2017 for writing 36-bit keys. Didn't get any today because my oscilloscope didn't have a "Trigger: Impulse"...

closed 2023-03-09 by team-orangeBlue