flipperdevices/flipperzero-firmware #2303: fbt can't seem to create a fap file

aunicorn111, 2023-01-14:
### Describe the bug.

Hi! I'm a newbie in creating custom applications for flipper.:)
My issue is, I get an error while compiling my code into a fap file. 

Simply restarting the compiling process didn't solve the issue. Not quite sure where I went wrong. Will appreciate any help! :))

### Reproduction

1. create a folder with appitself.c application.fam and a bitmap.png
2. run ./fbt fap_{your_app_id}
3. it burns:(

### Logs

```Text
./fbt fap_empty
	CC	applications_user/empty/empty.c
	PREGEN	build/f7-firmware-D/sdk_origin.i
	SDKSRC	build/f7-firmware-D/sdk_origin.i
	LINK	build/f7-firmware-D/.extapps/empty_d.elf
/home/irina/flipperzero-firmware/toolchain/x86_64-linux/bin/../lib/gcc/arm-none-eabi/10.3.1/../../../../arm-none-eabi/bin/ld: --gc-sections requires a defined symbol root specified by -e or -u
collect2: error: ld returned 1 exit status
scons: *** [build/f7-firmware-D/.extapps/empty_d.elf] Error 1

********** FBT ERRORS **********
build/f7-firmware-D/.extapps/empty_d.elf: Error 1
```

Milk-Cool, 2023-01-19:
same problem

Milk-Cool, 2023-01-19:
In my case, the solution was to rename to main function of the module. Its name was different from the one specified in `application.fam`

hedger, 2023-02-08:
That error message is usually caused by using wrong entry point name in your app manifest file. Please check that it points to a valid, defined function in your code.

Re-open issue if that is not the case.

closed 2023-02-08 by hedger