flipperdevices/flipperzero-firmware #2304: GPIO, UART bridge mode, high baud rates causes data loss

g3gg0, 2023-01-14:
### Describe the bug.

When reading data from a ESP32 based device that used 921600 baud, I basically got clean data.
But after some few dozen lines of startup log, the terminal didn't show any new data anymore.
First thinking this might be a terminal software issue, I tried switching to another terminal and realized the "missing" data arrived there when I rebooted the target.

Here a video where i 
 - start up the ESP32 based device
 - see "half" of the log
 - unpower the device
 - start PuTTY
 - power the device (causing new data to arrive)
 - suddenly see part of the missing data

https://cdn.discordapp.com/attachments/954430078882816021/1062478654183055420/hterm_2023-01-10_22-05-47.mp4

here another video where i
 - plug in the ESP32 based device
 - see "half" of the log
 - unplug the device
 - wiggle on the RX pin of the flipper to cause some data to arrive
 - suddenly parts of the missing data arrives

https://cdn.discordapp.com/attachments/954430078882816021/1062480901033971833/hterm_2023-01-10_22-18-37.mp4

Also some content of the log is missing from now and then.
Here a captured log, done with flipper:
[image]

and here when captured with a standard UART/USB adapter
[image]

So I see two issues that might exist
 a) the data is arriving too fast so the UART buffer overruns and data is lost
 b) some counter / ringbuffer pointers seem to be odd so that some data is stuck and is not immediately sent

### Reproduction

I am not sure if this is reproducible on your computer

here is what i've done to reproduce the issue.
 - attach an USB/UART adapter, wired up to the flippers RX/TX
 - use two terminals
 - connect one to flipper
 - connect the other to the USB/UART adapter
 - paste 1000 bytes into the terminal of the USB/UART adapter
 - no data seems to arrive until you e.g. press ENTER or some other immediately sent character a few times

Here a video:
https://user-images.githubusercontent.com/479231/212494003-a0b18008-52a5-4e40-8ebe-33001d63961c.mp4

DrZlo13, 2023-01-15:
Can you check if this issue is reproduced in dap link app?

g3gg0, 2023-01-17:
tried with this text:
[image]
[letter] [99 digits] [letter] [99 digits] ...

received using DAP-Link:
[image]

g3gg0, 2023-01-17:
it seems we are talking about 320 and 512 byte block sizes for GPIO app and DAP-Link app.

g3gg0, 2023-01-25:
@DrZlo13 any updates on this?

g3gg0, 2023-01-27:
Is this information enough and do you already know where to look?
Can you reproduce this or am I missing something or doing something wrong?

skotopes, 2023-02-10:
Okay, let's make one step back: Can you test devices separately? Get good USB serial adapter and check that flipper and esp behave correctly.

g3gg0, 2023-02-10:
as written in the first post:

USB-UART Adapter + ESP32 => good
[image]

Flipper + ESP32 => bad
[image]

g3gg0, 2023-02-13:
> Okay, let's make one step back: Can you test devices separately? Get good USB serial adapter and check that flipper and esp behave correctly.

does this match what you asked for?

g3gg0, 2023-02-13:
great, thanks

TQMatvey, 2023-03-07:
i was working on an esp32 raw monitor capturing mode, and used flipper to get data, it was also corrupred, baud rate was 921600, cant provide any more details as project is dropped due to those issues...

flipper definitely has issues capturing at high baud rates, not only in uart bridge, bad uart > sdcard as well as the app for flipper was written to directly try writing from uart to internal sdcard, worked but data was corrupted on high baud rates, 115200 was working perfectly

skotopes, 2024-07-07:
@g3gg0 should have been fixed in January UART refactoring.

skotopes, 2024-07-07:
reopen if issue persist.

closed 2024-07-07 by skotopes