flipperdevices/flipperzero-firmware #2312: Sub-GHz Incorrect range for IE (Ireland)

OzmoOzmo, 2023-01-17:
### Describe the bug.

Cannot transmit to 433 Mhz in Ireland using Sub-Ghz app.
Attempting to TX to 433.9Mhz
433-434 is permitted in EU (Ireland)
Flipper incorrectly says frequency is blocked for our region
Even though SubGhz Region menu lists this frequency as within legal ranges.

original firmware-Installed (0.75 ofw) as normal (no VPN used on installation pc)
See screenshots for details... It lists correct IE and the correct permitted ranges - but still will not TX on 433Mhz.

Lock Device we testing with is a unfitted model WIR222 receiver and 2 button key fob.
Thanks.
[image]
[image]
[image]
[image]
[image]

### Reproduction

1. Start SubGhz Read - Capture Key Fob
2. Replay Key Fob (with or without saving first)
3. Flipper says Blocked and will not TX.

### Target

A 2 Channel Receiver - identified as a "model WIR222" receiver and 2 button key fob.

### Logs

```Text
.sub file is as follows:

Filetype: Flipper SubGhz Key File
Version: 1
Frequency: 433920000
Preset: FuriHalSubGhzPresetOok650Async
Protocol: CAME
Bit: 40
Key: 00 00 00 53 0A FA AE EA
```

OzmoOzmo, 2023-01-17:
Some more info - hope it helps - log (from putty) - shows the following error
**[E][SubGhzProtocolCAME] Wrong number of bits in key**

From source code - it seems Flipper, when recording, is thinking this device is using "CAME" protocol - and when it sends - it realizes the bit count is not CAME compatible and then proceeds to give _the wrong error message_.

So issue is two fold - 
1. incorrectly detecting the protocol (should be RAW?)
2. showing error "Restricted in Region" when it should be "Cannot send due to bad Protocol/Corrupt save" or similar.

Reading and Replying as RAW worked! So my device has no rolling codes etc.

Skorpionm, 2023-01-17:
the protocol is not Came.  even if it looks a lot like him.  I did not specifically restrict the registration of such "similar" protocols, but prohibited the transfer.  post a photo of the remote control, receiver, what names are there, a photo of the board and, most importantly, RAW recordings of parcels.  5+ pieces of holding each of the remote control buttons for 3 seconds, and it would be nice if there were records of different remote controls

skotopes, 2023-02-10:
Yep, incorrect message that we showing to you. Issue with the key file. We'll update message.

skotopes, 2023-03-03:
Issue was fixed in #2443 please check and let us know if it still persist

closed 2023-03-03 by skotopes