flipperdevices/flipperzero-firmware #2313: Flipper crash with custom NotificationSequence

dcoles, 2023-01-17:
### Describe the bug.

When a user-defined `NotificationSequence` is run at the end of an external app using the non-blocking [`notification_sequence`](https://sourcegraph.com/github.com/flipperdevices/flipperzero-firmware@0.74.2/-/blob/applications/services/notification/notification.h?L87) API, the Flipper Zero will crash ~50% due to a Bus fault or NULL pointer dereference.

### Reproduction

External application used for reproduction (firmware: [0.74.2](https://github.com/flipperdevices/flipperzero-firmware/tree/0.74.2))

```c
#include <furi.h>
#include <notification/notification.h>
#include <notification/notification_messages.h>

const NotificationSequence chime = {
    &message_note_c4,
    &message_delay_100,
    &message_note_e4,
    &message_delay_100,
    &message_note_g4,
    &message_delay_100,
    &message_note_c5,
    &message_delay_100,
    &message_sound_off,
    NULL
};

int hello_world(char* arg) {
    (void) arg;
    
    NotificationApp* notifications = furi_record_open(RECORD_NOTIFICATION);
    notification_message(notifications, &chime);
    furi_record_close(RECORD_NOTIFICATION);

    return 0;
}
```

### Target

f7

### Logs

```Text
2783288 [I][fap_loader_app] FAP Loader is loading /ext/apps/Examples/hello_world.fap
2783338 [I][fap_loader_app] FAP Loader is mapping
2783388 [I][elf] Total size of loaded sections: 104
2783390 [I][fap_loader_app] Loaded in 102ms
2783393 [I][fap_loader_app] FAP Loader is starting app
2783407 [I][fap_loader_app] FAP app returned: 0
2783413 [D][BrowserWorker] Start
2783449 [D][BrowserWorker] Enter folder: /ext/apps/Examples items: 9 idx: 1
2783451 [D][BrowserWorker] Load offset: 0 cnt: 50
<Serial client disconnected>
```
```

skotopes, 2023-01-17:
That is expected behavior: notification_message is asynchronous function and if app exited and `sequence` is freed it will crash.

I'll update documentation and check what we can do to make it easier to use.

dcoles, 2023-01-19:
I was wondering if that was the case (particularly since `notification_message_block` exists).

Unfortunately, this makes `notification_message` tricky to use safely in external apps. A user would need to ensure their application waits if a notification message was still in progress. Not that it can't be done, but it's an easy mistake to make.

Out of curiosity, do notification sequences queue up? If so, could `notification_sequence_block` be used as a way to ensure all prior notification sequences are finished?

skotopes, 2023-01-19:
Main use case for notification_sequence_block is to wait till execution completed in a places where you may need it. And yes it can be used for that.

DrZlo13, 2023-01-25:
Fixed in #2313. Now you do not need a blocking notification when you exit the application.

PR #2335 (2023-01-25): ELF-loader: wait for notification to complete on app exit
# What's new

- Wait for notification to complete on app exit
- Fixes #2313

# Verification 

- #2313

# Checklist (For Reviewer)

- [x] PR has description of feature/bug or link to Confluence/Jira task
- [x] Description contains actions to verify feature/bugfix
- [x] I've built this code, uploaded it to the device and verified feature/bugfix
closed 2023-01-25 by dcoles