flipperdevices/flipperzero-firmware #2314: Support for analog input/output

Milk-Cool, 2023-01-17:
### Description of the feature you're suggesting.

Since a lot of GPIO sensors/input devices/etc. use analog input/output, it would be nice to have the Flipper Zero work with analog i/o out of box. Right now in the furi core libs there are no methods implementing this.

DrZlo13, 2023-02-11:
I have an [example with an analog input](https://github.com/DrZlo13/flipper-zero-adc-example), we plan to port it to the firmware.

sulph68, 2023-09-06:
hi

Just wondering if there has been any update on this capability? it would be really useful.

Thank you

skotopes, 2023-09-06:
@sulph68 available throw ST LL_HAL

khalifa007, 2023-12-10:
> @sulph68 available throw ST LL_HAL

Can u tell where ? Im trying to find the file

ecopsychologer, 2024-03-26:
> I have an [example with an analog input](https://github.com/DrZlo13/flipper-zero-adc-example), we plan to port it to the firmware.

Howdy! I’m trying to access the ADC and it seems like the firmware got rid of the access point you used in your example code, do you have any tips for accessing it now? Thanks so much!

MatthewKuKanich, 2024-04-01:
I'm also wondering. It seems things have changed some since your example app. Thanks!

Milk-Cool, 2024-04-06:
> Howdy! I’m trying to access the ADC and it seems like the firmware got rid of the access point you used in your example code, do you have any tips for accessing it now? Thanks so much!

@ecopsychologer it looks like the stm libraries are now inaccessible for flipper zero applications. someone needs to at least make a wrapper for adc functions that would be accessible from an app.

ecopsychologer, 2024-04-06:
> @ecopsychologer it looks like the stm libraries are now inaccessible for flipper zero applications. someone needs to at least make a wrapper for adc functions that would be accessible from an app.

Ahhh that makes so much sense! I bought myself a similar board to what the flipper was built on, and am now having to navigate the the adc/hal functions and such anyways on that side, so if I figure anything out I'll be sure to pass it along and disseminate that info as widely as possible

ecopsychologer, 2024-04-06:
My current workaround has been using the rogue master firmware which it seems did not quite yet phase out adc support. I don't really know how to build for specific firmwares using ufbt but I’m pretty sure there's a way. That may be good enough until some folx savvier than I bring the lower level functionality back to the firmware

ecopsychologer, 2024-04-06:
@Milk-Cool Im realizing now in my research into creating wrappers as you mentioned (I’m an amateur) that it's very possible that what I saw in the flipperscope repo (and application but if you search that you'll find the source code) is in fact the wrapper for the hardware level libraries. If you look into that project, it has folders with different files with libraries that it seems the author uses to access adc. My first attempt to understand it wasn't successful but maybe two brains are better than one? I'd be down to discuss on a forum or discord or somewhere that isn't gonna ping the lovely folx working on the firmware. 

Thoughts?
Best,
J

Milk-Cool, 2024-04-06:
> @Milk-Cool Im realizing now in my research into creating wrappers as you mentioned (I’m an amateur) that it's very possible that what I saw in the flipperscope repo (and application but if you [...]

sorry, don't have a flipper with me rn :( so no way to test anything

ecopsychologer, 2024-04-06:
@Milk-Cool thats all good, I made a repo called flipper-zero-adc-wrapper where I'll be trying to figure stuff out. Regardless of your ability to test, I'd definitely appreciate any help or ideas that come to you if you wanted to give it a glance. It's early stages but I’m hoping it'll become something handy for folx like us haha

skotopes, 2024-04-14:
https://github.com/flipperdevices/flipperzero-firmware/pull/3583 coming next release

ecopsychologer, 2024-04-14:
> https://github.com/flipperdevices/flipperzero-firmware/pull/3583 coming next release

Bless you! Thanks!

PR #3583 (2024-04-17): [FL-2969] FuriHal: add ADC API
# What's new

- FuriHal: add ADC API
- fix #2314

# Verification 

- Check example_adc

# Checklist (For Reviewer)

- [x] PR has description of feature/bug or link to Confluence/Jira task
- [x] Description contains actions to verify feature/bugfix
- [x] I've built this code, uploaded it to the device and verified feature/bugfix
closed 2024-04-17 by skotopes