flipperdevices/flipperzero-firmware #2328: nfc can't save cards

littleblack111, 2023-01-23:
### Description of the feature you're suggesting.

please add reading, saving, emulating ALL cards back(the more options)

### Anything else?

idk why is it removed, it should be very easy to undo those commits(a lot of them, can't list all) and i don't understand what does the "saved" do, if you can't save it, then why do you need that??

skotopes, 2023-02-10:
Which specific card type we are talking about?

Ethanol0001, 2023-02-28:
keep in mind if you are trying to do bank cards, the flipper wont let you save them. Have you tried something like an Amiibo?

Astrrra, 2023-03-07:
Saving and Emulating bank cards will not be added back, as it had never worked, and never will. You wouldn't be able to use the saved bank card dump for payment, as it requires non-extractable keys. Therefore, the emulation feature was just confusing people (as it never worked), and was removed. We do not plan on adding that back, as you physically cannot emulate those cards.

closed 2023-03-07 by Astrrra