flipperdevices/flipperzero-firmware #2339: DESfire ID emulation no signal

GitGitBunny, 2023-01-26:
### Describe the bug.

Ready for a headache? This is not for the faint of heart, an absolute challenge. 

Yesterday morning i used my Flipper to read DESfire cards and emulate them to open a few locks during an approved penetration assessment. During the simulation, the Flipper froze during the emulation of the ID so after every emulation i had to hard reset the Flipper (left+back).

In the afternoon emulation of Desfire cards stopped entirely. I have a working NFC card (employee badge) that i tried to emulate and that has been working but now doesn't anymore. The reader does not recognize any signal coming from my Flipper anymore! Neither does my iPhone, when simulating the UID of Desfire (ATQA 4403 or 0403 (last of which is random UID)). Any other NFC card works, it is only when simulating ATQA4403 or 0403 with SAK20. Trying to simulate NFC-A 7 byte with ATQA0400 SAK88 works fine. 

I tried manually editing in all different variants that can be manually added, emulation of all work, all except for Desfire. Reading of the Desfire card does work without a problem. It registers properly and identifies it as a DESfire card, but when emulating, it seems no signal is received. When emulating, the UID shows, the led is blinking purple, but the NFC reader does not receive any signal. 

One more weird thing: when i just open my iPhone's NFC menu and scan my tag, the Flipper shows a 'log' button and when i click it i get the same message over and over again 'R: 00' whenever i hold the Flipper close to the reader. The weird thing: no debug logs register. 

### Reproduction

1. Go To NFC
2. Read Card
3. Emulate ID
4. No response by reader.

### Target

iPhone NFC reader

### Logs

```Text
# Reading DESfire (twice) and emulating the second time. 
291156 [D][DolphinState] icounter 20, butthurt 0
294433 [D][ViewDispatcher] View changed while key press 2000E8E8 -> 2000E130. Sending key: Back, type: Release, sequence: 00000032 to previous view port
295807 [D][ViewDispatcher] View changed while key press 2000E130 -> 2000E8E8. Sending key: Back, type: Release, sequence: 00000033 to previous view port
297271 [D][ViewDispatcher] View changed while key press 2000E8E8 -> 2000E208. Sending key: Back, type: Release, sequence: 00000034 to previous view port
303573 [D][DolphinState] icounter 20, butthurt 0
304151 [I][NfcWorker] Mifare DESFire detected
304374 [W][MifareDESFire] Bad DESFire GET_KEY_SETTINGS response
304605 [D][DolphinState] icounter 20, butthurt 0
325534 [I][NfcWorker] Mifare DESFire detected
325755 [W][MifareDESFire] Bad DESFire GET_KEY_SETTINGS response
325982 [D][DolphinState] icounter 20, butthurt 0
330933 [D][DolphinState] icounter 20, butthurt 0
336857 [D][ViewDispatcher] View changed while key press 2000E8E8 -> 2000E130. Sending key: Back, type: Release, sequence: 0000003D to previous view port

#Scan, save and emulate
391617 [D][DolphinState] icounter 20, butthurt 0
392606 [I][NfcWorker] Mifare DESFire detected
392825 [W][MifareDESFire] Bad DESFire GET_KEY_SETTINGS response
393057 [D][DolphinState] icounter 20, butthurt 0
401131 [I][Nfc] Saving to folder /any/nfc
402759 [D][BrowserWorker] Start
402777 [D][BrowserWorker] Enter folder: /any/nfc items: 7 idx: 6
402780 [D][BrowserWorker] Load offset: 0 cnt: 50
406241 [D][BrowserWorker] End
406343 [W][NfcDevice] Non-parsed apps found!
409420 [D][DolphinState] icounter 20, butthurt 0

#Emulating Mifare classic on my iPhone when not waiting for NFC input: 
1508177 [D][BrowserWorker] Start
1508189 [D][BrowserWorker] Enter folder: /any/nfc items: 7 idx: -1
1508193 [D][BrowserWorker] Load offset: 0 cnt: 50
1514436 [D][BrowserWorker] End
1515376 [D][DolphinState] icounter 20, butthurt 0
1517712 [D][MfClassic] 0e56deb5 keyA block 3 nt/nr/ar: 3277a1a5 d805f324 fca4b1cb
1517771 [W][MfClassic] Incorrect nr + ar
1517822 [D][MfClassic] 0e56deb5 keyA block 3 nt/nr/ar: 12817c7f 3423dd49 75390f7c
1517877 [D][MfClassic] 0e56deb5 keyA block 3 nt/nr/ar: c4fafb0e 4f298bef a1c91059
1517931 [D][MfClassic] 0e56deb5 keyA block 3 nt/nr/ar: 50f3745c 26901b9f 29824fbf
1517985 [D][MfClassic] 0e56deb5 keyA block 3 nt/nr/ar: 1bcf4639 fbd7f740 5ebc98f0
1518040 [D][MfClassic] 0e56deb5 keyA block 3 nt/nr/ar: 17acf0bd f84e0a3f 9970f17a
1524399 [D][ViewDispatcher] View changed while key press 2000EB60 -> 2000E980. Sending key: Back, type: Release, sequence: 0000005C to previous view port

#Emulating DESfire on my Iphone while it is not waiting for NFC input (see photo for the R: 00 message)
1528508 [D][BrowserWorker] Start
1528528 [D][BrowserWorker] Enter folder: /any/nfc items: 7 idx: 3
1528530 [D][BrowserWorker] Load offset: 0 cnt: 50
1533252 [D][BrowserWorker] End
1533348 [W][NfcDevice] Non-parsed apps found!
1534297 [D][DolphinState] icounter 20, butthurt 0
1564304 [I][Dolphin] Flush stats
1564306 [I][SavedStruct] Saving "/int/.dolphin.state"
1564317 [D][StorageInt] Device erase: page 3, translated page: c0
1564324 [D][StorageInt] Device sync: skipping
1564327 [I][DolphinState] State saved
1569827 [D][ViewDispatcher] View changed while key press 2000F0D0 -> 2000F1E8. Sending key: Back, type: Release, sequence: 00000066 to previous view port
1571100 [D][ViewDispatcher] View changed while key press 2000F1E8 -> 2000E980. Sending key: Back, type: Release, sequence: 00000067 to previous view port
```

### Anything else?

To be clear: this has worked. It was perfectly fine and working yesterday morning and now it does not anymore. Below is the R:00 message on the flipper. Do you need more info? please request i'd be happy to share and participate in debugging this issue together. 

[image]

gornekich, 2023-01-26:
Hello @GitGitBunny 

When you emulate DESFire UID, after anticollision reader sends RATS and tag respond with ATS. The reason why everything works except DESFire UID emulation is that Flipper doesn't respond with ATS.

I know about this bug and I was sure that I fixed that in #2098. What firmware version do you use?

GitGitBunny, 2023-01-26:
Thanks for the quick Reply @gornekich. 

Some more details: 
HW: 12.F7B9C6 R04:NL Lami
FW: 0.75.0 [08-01-2023] 8ee66c3e [1854] 1.13.3:L [7] 0.75.0

GitGitBunny, 2023-01-27:
@gornekich in addition, this just hit me, i just want to confirm you that it was indeed working so i think you actually fixed the problem. However, it stopped working all of a sudden after a couple of uses and freezes and a firmware repair does not seem to fix the problem. To your expert knowledge, maybe this could be a hardware issue? that it somehow broke down because it froze a couple of times? Is there a way i can provide you with test data if the hardware is functioning properly? i'd be happy to ablige.

gornekich, 2023-01-27:
@GitGitBunny I think it's not a hardware issue, since other cards emulation works. Could you read DESFire emulation with reader, which can show ATS? I want to be sure that ATS is sent after anticollision. It may be incorrect, but if your tag is iso14443-4 complient (5 bit in SAK is 1), then reader sends RATS and tag must answer with ATS.

The easiest way to do this for me is to read flipper emulation with Android phone with NFC tool application

GitGitBunny, 2023-02-15:
@gornekich i have done further testing at home. Whenever i am emulating the DESFire card the reader receives no signal. Also when manually transmitting a NFC-A 7-Bytes UID with SAK20 ATQA 0344 the reader receives no signal. I've spoken with several colleagues they suggested the terminals i have used them on maybe were in some sort of dual mode when i tried to open them and by coincidence they disabled that at that day. 

The only trail i have is that the Flipper is sending some kind of signal or receiving some kind of signal because of the R:00 error message that is logged on the Flipper. But there is no transmission of data or a handshake. 

I have read online some info on bricking magic cards with a bad BCC value. Could that be the case? That i accidently entered wrong values and my FZ is bricked for DESfire communication?

itsomarawad, 2023-04-13:
Unfortunately also having the same issue. Has there been any progress on this? Bit of a bummer since I just received my Zero. Also tried a DFU reset but to no avail.

GitGitBunny, 2023-05-05:
@666badguy666 @itsomarawad sorry for the late response, been away for work and couldn't find the time to respond. 

I've went to the bottom of this issue, in my case it was a situation of circumstance. The readers that previously did read the emulation were updated that same day to no longer support UID only support. Emulation does work, but DESFire just seems to be very unsupported by the FZ, which is becoming more common by the day. Other emulation does work. This issue makes the FZ irrelevant when compared to products such as Lexmark, but it is still the best tool for SubGHZ. Hope that helps, just don't expect to much regarding NFC simulation.

doomwastaken, 2023-11-03:
@GitGitBunny @itsomarawad could you please verify your issue on latest firmware? There was entire NFC refactor

doomwastaken, 2023-11-21:
Please reopen if issue persists

closed 2023-11-21 by doomwastaken