flipperdevices/flipperzero-firmware #2345: Sub Ghz cannot parse file/Transmission Blocked

kritonpc, 2023-01-29:
### Describe the bug.

Although I've never had any issues with sub ghz I encountered an issue some time in the past and again today.
I recorded a remote that works at 433.92 and when i pressed send directly after recording it and it was recognized, it showed me a transmission error saying that the specific frequency is not supported (although it is 433.92, which is supported in my country). I saved it and tried sending it and when selecting the file it shows the error "Cannot parse file". I tried it several times but it doesn't work. When saving it RAW of course it worked, but the rest seems odd. Any idea what can be wrong? I tried it on both OFW and on a CFW with unlocked transmission and it persists. Here are the .sub files of the 2 remotes as they were saved on flipper:

```
Filetype: Flipper SubGhz Key File
Version: 1
Frequency: 433920000
Preset: FuriHalSubGhzPresetOok650Async
Protocol: CAME
Bit: 32
Key: 00 00 00 00 66 CF 63 02
```
and
```
Filetype: Flipper SubGhz Key File
Version: 1
Frequency: 433889000
Preset: FuriHalSubGhzPresetOok650Async
Protocol: CAME
Bit: 32
Key: 00 00 00 00 F7 46 88 12

```

### Reproduction

1. Record the remote 
2. Press send directly after recording it 
3. It shows a transmission error saying that the specific frequency is not supported

and 
1. Save the remote as a .sub file
2. Select it
3. It shows the error "Cannot parse file"

Skorpionm, 2023-02-02:
it's not a bug, came.  The protocol only supports 12 and 24 bits.  instead of the other varieties of this protocol is not the same.  send raw.  recording several long presses on each button on the remote control.  and attach a photo and the name of your system, I will add it

kritonpc, 2023-02-03:
This is for the first one
```Filetype: Flipper SubGhz RAW File
Version: 1
Frequency: 433920000
Preset: FuriHalSubGhzPresetOok650Async
Protocol: RAW
RAW_Data: -1154 163 -296 165 -990 99 -132 365 -198 199 -364 925 -432 99 865 51303 -14132 65 -2054 65 -100 393 -860 489 -100 163 -562 789 -98 165 -64 37125 -13712 197 -132 97 -1858 895 -332 101 -398 993 -164 267 -200 232083 -6698 65 -5076 67 -364 331 -168 231 -230 395 -990 495 -766 165 -66 859 -660 885 -100 327 -230 31739 -664 273 -386 583 -682 277 -384 559 -682 279 -388 553 -698 275 -386 557 -376 603 -356 589 -386 587 -358 579 -380 585 -372 575 -684 277 -690 307 -660 279 -360 591 -682 279 -356 623 -664 275 -664 303 -680 255 -696 283 -660 289 -680 277 -356 591 -374 587 -684 275 -686 273 -660 311 -358 593 -11474 317 -362 593 -352 607 -656 289 -682 313 -630 305 -652 311 -660 311 -324 603 -678 283 -362 597 -320 637 -668 291 -328 637 -642 319 -632 295 -672 315 -340 607 -344 631 -644 315 -662 279 -662 315 -310 635 -668 301 -660 283 -344 601 -672 283 -338 633 -668 293 -660 283 -336 631 -672 289 -328 645 -11472 305 -350 601 -644 319 -666 289 -344 615 -346 623 -644 311 -660 275 -356 605 -676 283 -654 291 -340 647 -314 619 -672 311 -630 305 -648 317 -662 319 -318 621 -672 301 -662 283 -342 601 -376 603 -320 629 -672 289 -658 283 -338 631 -354 619 -328 615 -346 623 -344 609 -326 609 -656 291 -340 647 -11466 329 -638 309 -662 315 -324 631 -648 319 -326 589 -658 317 -634 331 -312 645 -326 607 -658 291 -666 315 -636 329 -642 313 -626 343 -622 309 -662 313 -662 309 -626 319 -362 595 -644 319 -326 625 -324 651 -622 321 -658 311 -628 307 -356 605 -646 317 -656 309 -304 647 -324 619 -652 309 -308 653 -11450 343 -656 323 -326 595 -680 285 -668 287 -344 615 -654 305 -652 311 -630 345 -322 607 -324 649 -322 623 -652 317 -638 291 -670 311 -628 307 -356 639 -642 321 -326 625 -646 309 -630 317 -364 595 -320 631 -658 289 -350 613 -670 313 -620 345 -304 643 -626 321 -308 653 -314 621 -672 313 -626 307 -11478 317 -362 589 -674 313 -628 307 -650 315 -660 283 -362 589 -360 615 -660 299 -648 299 -658 321 -306 633 -644 315 -306 635 -344 635 -322 633 -318 663 -314 621 -672 313 -628 307 -652 311 -660 311 -622 297 -336 633 -346 631 -308 637 -318 631 -348 597 -672 283 -660 319 -662 299 -346 611 -646 313 -11446 353 -628 307 -652 311 -660 311 -326 603 -322 653 -658 287 -344 621 -350 619 -310 613 -652 345 -324 629 -296 635 -648 321 -634 313 -326 625 -672 313 -658 315 -624 297 -658 319 -326 623 -326 649 -314 623 -338 627 -656 303
RAW_Data: -650 311 -308 635 -356 591 -352 609 -352 621 -332 615 -334 633 -320 633 -11478 323 -630 331 -640 285 -336 637 -668 299 -644 283 -336 635 -670 289 -660 283 -338 631 -672 289 -660 311 -626 311 -356 595 -346 625 -648 311 -308 667 -658 289 -672 281 -338 629 -658 289 -678 279 -666 301 -648 311 -308 637 -656 291 -338 649 -630 297 -674 313 -626 307 -656 309 -664 311 -626 297 -11484 321 -670 289 -328 635 -642 319 -338 601 -352 599 -354 605 -346 627 -346 601 -318 635 -346 631 -310 637 -320 631 -670 291 -328 615 -332 635 -656 323 -638 313 -336 629 -658 289 -346 615 -346 623 -646 313 -310 633 -656 291 -662 311 -660 321 -328 595 -318 635 -354 619 -630 315 -362 589 -338 647 -11438 335 -350 605 -646 317 -328 633 -644 321 -306 631 -350 599 -648 317 -662 301 -646 315 -660 275 -662 315 -662 309 -622 319 -326 635 -642 319 -326 625 -360 617 -656 291 -332 615 -650 343 -324 627 -648 311 -624 309 -342 605 -356 623 -320 611 -660 321 -650 283 -336 633 -658 289 -676 281 -660 319 -11472 309 -658 319 -326 627 -648 321 -632 289 -346 617 -346 623 -312 647 -326 607 -356 625 -318 611 -352 619 -666 315 -628 303 -352 607 -326 645 -648 317 -652 283 -338 631 -658 289 -678 281 -338 629 -658 289 -348 615 -654 307 -652 311 -662 311 -324 605 -644 319 -326 635 -642 319 -326 625 -322 649 -11450 319 -336 645 -650 311 -626 297 -660 319 -306 633 -350 599 -678 319 -326 623 -626 315 -636 329 -312 645 -646 315 -620 345 -302 645 -648 287 -676 313 -308 635 -646 319 -668 283 -660 321 -326 627 -646 321 -632 289 -644 349 -326 589 -340 647 -296 637 -656 291 -338 645 -624 321 -660 311 -310 635 -11480 313 -658 315 -652 285 -660 319 -326 619 -630 313 -668 301 -646 285 -660 319 -326 623 -654 317 -326 621 -334 615 -173470 265 -502 131 -166 593 -164 263 -950 331 -98 299 -766 825 -298 163 -496 1283 -130 295 -12142 397 -1016 465 -200 197 -916 331 -1350 425 -100 327 -924 961 -100 165 -100 133 -400 1091 -66 197 -98 20515 -10968 97 -1026 229 -164 97 -130 291 -764 399 -232 297 -1064 97 -134 459 -296 165 -528 887 -268 1659 -100 299 -100 8081 -10714 65 -1654 99 -3016 165 -794 65 -532 799 -68 133 -696 1191 -66 399 -200 14585 -7520 97 -3066 97 -64 293 -1654 197 -100 991 -392 795 -66 299 -264
```

This is for the second
```
Filetype: Flipper SubGhz RAW File
Version: 1
Frequency: 433889000
Preset: FuriHalSubGhzPresetOok650Async
Protocol: RAW
RAW_Data: 23879 -66 10887 -16830 199 -200 65 -232 65 -894 927 -766 331 -364 697 -68 3815 -132 40451 -100 14373 -132 21721 -66 3613 -100 4577 -100 335 -66 4783 -66 2291 -132 2987 -132 16599 -16668 65 -764 131 -132 99 -566 65 -232 593 -266 299 -498 165 -166 1493 -132 13829 -66 3453 -132 2325 -17520 263 -134 99 -234 299 -98 697 -1020 857 -66 131 -134 235 -200 65 -98 2707 -66 1599 -102 3857 -66 1885 -66 1031 -100 1793 -100 5639 -66 1663 -100 5289 -98 1915 -66 1559 -98 5407 -166 1261 -66 3329 -102 3733 -66 3235 -130 1675 -100 3823 -100 4655 -66 4985 -132 19641 -130 229 -66 4607 -100 9971 -16444 63 -1558 65 -100 231 -1030 363 -828 1493 -68 1573 -132 295 -98 1237 -132 301 -102 3659 -132 4759 -1 [... cut ...]  1865 -132 2041 -66 1349 -98 623 -100 1557 -66 10817 -100 5383 -134 1497 -98 3125 -66 3349 -134 365 -134 2897 -98 20625 -17492 165 -198 431 -100 65 -398 165 -98 99 -100 599 -856 623 -100 389 -130 325 -100 11797 -98 3989 -68 12123 -100 10501 -102 199 -134 21763 -66 1341 -66 12767 -100 8047 -17524 97 -294 97 -132 923 -900 861 -66 497 -66 16061 -66 2989 -98 651 -66 6905 -100 3613
```

This is all I have, as they are from friends and family. I have to get extra info from them. What else do you need?

Skorpionm, 2023-02-05:
still need the name of the system and a photo of the remote control

kritonpc, 2023-02-06:
For the first one, it seems to be a Proteco Mover 5 or something. I couldn't find information on it, but google lens seemed to do the job.
[image]
This is the motor and it seems to be this https://www.adg.ie/Automation/mover_mag--Proteco-Mover-Sliding-Gate-Motor-Kit-

The remote is this:
[image]
[image]

I'll try to find the second one ASAP. If you need additional information let me know

skotopes, 2023-03-03:
Issue was fixed in #2443 please check and let us know if it still persist

closed 2023-03-03 by skotopes

kritonpc, 2023-03-10:
It still doesn't allow me to transmit it. I just get a more specific error.

kritonpc, 2023-03-10:
"Error in protocol parameters description"

Skorpionm, 2023-03-10:
now you can include BinRaw in the config, and accept this console again. should persist and correctly play unsupported protocols

kritonpc, 2023-03-10:
@Skorpionm What do you mean? What should I change on the saved file? What actions should i follow?

Skorpionm, 2023-03-11:
no need to do anything with the saved file.  open read -> config -> bin raw .  return to read and accept the signal from the remote control again, you will have a binraw format signal.  save this and use it, it should repeat the signal from your key fob in strength.  also works, transmits while the transmit button is pressed

dkorunic, 2024-01-30:
The system above is Proteco MOVER, I can confirm. Can we get a protocol parser for this @Skorpionm please? It's quite common 230V gate mover/opener system in EU and produced in Italy. I'm able to send a few raw files if needed.

Fellahosting, 2025-04-07:
same error BIN RAW TRANSMISSION IS BLOCKED 

[image]

dkorunic, 2026-11-09:
I've written a very simple Proteco garage opener app that works for me and I'll try to get protocol patches accepted if anyone is still interested: https://github.com/dkorunic/flipper-proteco