flipperdevices/flipperzero-firmware #2351: Regression with PR #2210

adryd325, 2023-02-01:
### Describe the bug.

This PR (#2210) seems to have broken the ability for me to control my ceiling fan

Before this was merged controlling my fan worked perfectly; however after this commit I have to hold down send for several seconds to get my ceiling fan to respond.

If needed I can take I/Q recordings of the original transmission, the transmission before this update, and the transmission after this update.

### Reproduction

1. Record a CAME transmission
2. Attempt to replay it
3. It requires many multiple replays before the appliance reacts

Expected result: a short press should cause the appliance to react

Skorpionm, 2023-02-01:
you mean not CAME? but a similar protocol. that PR was made from the original CAME console. record Read RAW for 5+ clicks of each button on your remote. attach here with a photo of the remote control itself

adryd325, 2023-02-06:
[fan_on_hi_raw.sub.txt](https://github.com/flipperdevices/flipperzero-firmware/files/10669144/fan_on_hi_raw.sub.txt)

I'll get a photo in a bit, but for now, it seems to be a CHQ9051T or a clone

tox16b, 2023-02-23:
I also noticed this regression, I have a CAME gate which stopped working after 0.75 update, and still doesn't work with 0.77.1.
I captured raw signals from my flipper using another flipper, if I playback with 0.74 the receiving flipper recognizes it as CAME 70, but if I playback with 0.77.1 it doesn't recognize it. I made a raw capture of both signals.
[0.74.sub.txt](https://github.com/flipperdevices/flipperzero-firmware/files/10817480/0.74.sub.txt)
[0.77.1.sub.txt](https://github.com/flipperdevices/flipperzero-firmware/files/10817481/0.77.1.sub.txt)

adryd325, 2023-02-26:
@tox16b maybe a photo of the remote would help them too

skotopes, 2023-03-03:
Issue was fixed in #2443 please check and let us know if it still persist

closed 2023-03-03 by skotopes