flipperdevices/flipperzero-firmware #2355: Odd behavior NFC Read with clone card and cached keys

ML7-Secure, 2023-02-02:
### Describe the bug.

I have a Mifare Classic 1K card saved
I can’t get 3 keys and so I can’t read the last 3 sectors correctly

I cloned this partial card (on a magic card), I read the clone and saved this clone too

When I scanned the **original** card another time it « found » the 3 remaining keys putting null 0x00 keys as valid keys but still the sectors couldn’t be read, so these last « found » keys seem incorrect, they seem to be keys of the blank card

After erasing cache no more problem, when I scan back the original card, the (incorrect) null keys are no more marked as correct (as it goes through all the dictionary another time)

### Reproduction

1. Read an (original) Mifare Classic card which has some keys that are not in the dict 
2. Clone this original card
3. Read the clone and save it
4. Read the original card (all keys should be found, _the problem seems to be here_)
5. Erase cached keys
6. Read the original card (unknown keys should not be found)

skotopes, 2023-02-10:
Can you attach device logs for step 4?

ML7-Secure, 2023-02-10:
Sectors 13, 14, 15 have keys that are not in dict

Logs for step 3 (read clone) : 
```
[I][NfcWorker] Mifare Classic detected
[I][NfcWorker] Trying to read a supported card ...
[D][Plant] Verifying sector 8
[D][FuriHalNfc] Timeout during data exchange
[D][Troika] Verifying sector 11
[D][FuriHalNfc] Timeout during data exchange
[I][NfcWorker] Search for key cache ...
[I][NfcWorker] Load keys cache success. Start reading
[D][MfClassic] Try to read blocks with key A
[D][MfClassic] Read 4 blocks out of 4
[D][MfClassic] Try to read blocks with key A
[D][MfClassic] Read 4 blocks out of 4
[D][MfClassic] Try to read blocks with key A
[D][MfClassic] Read 4 blocks out of 4
[D][MfClassic] Try to read blocks with key A
[D][MfClassic] Read 4 blocks out of 4
[D][MfClassic] Try to read blocks with key A
[D][MfClassic] Read 4 blocks out of 4
[D][MfClassic] Try to read blocks with key A
[D][MfClassic] Read 4 blocks out of 4
[D][MfClassic] Try to read blocks with key A
[D][MfClassic] Read 4 blocks out of 4
[D][MfClassic] Try to read blocks with key A
[D][MfClassic] Read 4 blocks out of 4
[D][MfClassic] Try to read blocks with key A
[D][MfClassic] Read 4 blocks out of 4
[D][MfClassic] Try to read blocks with key A
[D][MfClassic] Read 4 blocks out of 4
[D][MfClassic] Try to read blocks with key A
[D][MfClassic] Read 4 blocks out of 4
[D][MfClassic] Try to read blocks with key A
[D][MfClassic] Read 4 blocks out of 4
[D][MfClassic] Try to read blocks with key A
[D][MfClassic] Read 4 blocks out of 4
[D][MfClassic] Try to read blocks with key A
[D][FuriHalNfc] Timeout during data exchange
[D][FuriHalNfc] Timeout during data exchange
[D][FuriHalNfc] Timeout during data exchange
[D][MfClassic] Read 0 blocks out of 4
[D][MfClassic] Try to read blocks with key A
[D][FuriHalNfc] Timeout during data exchange
[D][FuriHalNfc] Timeout during data exchange
[D][FuriHalNfc] Timeout during data exchange
[D][MfClassic] Read 0 blocks out of 4
[D][MfClassic] Try to read blocks with key A
[D][FuriHalNfc] Timeout during data exchange
[D][FuriHalNfc] Timeout during data exchange
[D][FuriHalNfc] Timeout during data exchange
[D][MfClassic] Read 0 blocks out of 4
[D][MfClassic] Read 13 sectors and 29 keys
[I][NfcWorker] Read 13 sectors out of 16 total
[I][MfClassicDict] Loaded dictionary with 0 keys
[D][NfcWorker] Start Dictionary attack, Key Count 0
[I][NfcWorker] Sector 0
[I][NfcWorker] Sector 1
[I][NfcWorker] Sector 2
[I][NfcWorker] Sector 3
[I][NfcWorker] Sector 4
[I][NfcWorker] Sector 5
[I][NfcWorker] Sector 6
[I][NfcWorker] Sector 7
[I][NfcWorker] Sector 8
[I][NfcWorker] Sector 9
[I][NfcWorker] Sector 10
[I][NfcWorker] Sector 11
[I][NfcWorker] Sector 12
[... 212 lines cut ...]
[D][NfcWorker] Try to auth to sector 13 with key ffffffffffff
[D][MfClassic] Try to read blocks with key A
[D][FuriHalNfc] Timeout during data exchange
[D][FuriHalNfc] Timeout during data exchange
[D][MfClassic] Read 1 blocks out of 4
[D][MfClassic] Try to read blocks with key B
[D][FuriHalNfc] Timeout during data exchange
[I][NfcWorker] Sector 14
[D][NfcWorker] Try to auth to sector 14 with key ffffffffffff
[D][MfClassic] Try to read blocks with key A
[D][FuriHalNfc] Timeout during data exchange
[D][FuriHalNfc] Timeout during data exchange
[D][FuriHalNfc] Timeout during data exchange
[D][MfClassic] Read 0 blocks out of 4
[D][MfClassic] Try to read blocks with key B
[D][FuriHalNfc] Timeout during data exchange
[I][NfcWorker] Sector 15
[D][NfcWorker] Try to auth to sector 15 with key ffffffffffff
[D][MfClassic] Try to read blocks with key A
[D][FuriHalNfc] Timeout during data exchange
[D][FuriHalNfc] Timeout during data exchange
[D][FuriHalNfc] Timeout during data exchange
[D][MfClassic] Read 0 blocks out 
[D][MfClassic] Try to read blocks with key B
```

ML7-Secure, 2023-03-18:
tested with [7de7fa29](https://github.com/flipperdevices/flipperzero-firmware/commit/7de7fa293b6eb8def92ecfecc1c1aea63a269624), no more problem, good job

closed 2023-03-18 by ML7-Secure

Karl99999, 2023-03-18:
I copied a mifare tag, from my Door key, and it’s working fantastic, but how can I write it to a new plastic tag, I bought many new tags in China, but the write function in flipper zero isn’t working, I can’t only use the flipper device and it’s not possible to write the mifare tag to a new tag that is writable, and rewritable!! Can someone help me with that?

Filetype: Flipper NFC device
Version: 3
# Nfc device type can be UID, Mifare Ultralight, Mifare Classic
Device type: Mifare Classic
# UID, ATQA and SAK are common for all formats
UID: 04 BC 9B 82 93 2F 80
ATQA: 00 44
SAK: 08
# Mifare Classic specific data
Mifare Classic type: 1K
Data format version: 2
# Mifare Classic blocks, '??' means unknown data
Block 0: 04 BC 9B 82 93 2F 80 88 44 00 C2 00 00 00 00 00
Block 1: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
Block 2: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
Block 3: 12 FE ED 78 BE EF FF 07 80 69 ?? ?? ?? ?? ?? ??
Block 4: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
Block 5: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
Block 6: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
Block 7: FF FF FF FF FF FF FF 07 80 69 FF FF FF FF FF FF
Block 8: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
Block 9: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
Block 10: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
Block 11: FF FF FF FF FF FF FF 07 80 69 FF FF FF FF FF FF
Block 12: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
Block 13: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
Block 14: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
Block 15: FF FF FF FF FF FF FF 07 80 69 FF FF FF FF FF FF
Block 16: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
Block 17: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
Block 18: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
Block 19: FF FF FF FF FF FF FF 07 80 69 FF FF FF FF FF FF
Block 20: 22 70 85 DF 01 00 00 00 0A 0F 00 38 6D FD 69 7A
Block 21: 43 E5 69 00 00 00 00 00 00 00 00 00 00 00 00 00
Block 22: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
Block 23: DC FB E0 D3 2C CF FF 07 80 69 FF FF FF FF FF FF
Block 24: 00 00 10 40 00 00 00 00 00 00 00 00 00 07 00 00
Block 25: 04 83 6A 80 00 00 00 00 00 00 00 00 00 00 00 00
Block 26: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
Block 27: DC FB E0 D3 2C CF FF 07 80 69 FF FF FF FF FF FF
Block 28: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
Block 29: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
Block 30: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
Block 31: DC FB E0 D3 2C CF FF 07 80 69 FF FF FF FF FF FF
Block 32: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
Block 33: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
Block 34: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
Block 35: DC FB E0 D3 2C CF FF 07 80 69 FF FF FF FF FF FF
Block 36: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
Block 37: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
Block 38: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
Block 39: DC FB E0 D3 2C CF FF 07 80 69 FF FF FF FF FF FF
Block 40: E8 E5 DB 8C 01 00 00 00 00 00 00 00 00 00 00 00
Block 41: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
Block 42: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
Block 43: DC FB E0 D3 2C CF FF 07 80 69 FF FF FF FF FF FF
Block 44: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
Block 45: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
Block 46: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
Block 47: DC FB E0 D3 2C CF FF 07 80 69 FF FF FF FF FF FF
Block 48: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
Block 49: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
Block 50: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
Block 51: DC FB E0 D3 2C CF FF 07 80 69 FF FF FF FF FF FF
Block 52: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
Block 53: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
Block 54: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
Block 55: FF FF FF FF FF FF FF 07 80 69 FF FF FF FF FF FF
Block 56: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
Block 57: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
Block 58: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
Block 59: FF FF FF FF FF FF FF 07 80 69 FF FF FF FF FF FF
Block 60: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
Block 61: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
Block 62: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
Block 63: FF FF FF FF FF FF FF 07 80 69 FF FF FF FF FF FF

ML7-Secure, 2023-03-18:
@Karl99999 please visit the [official FZ Discord](http://flipperzero.one/discord) or the [official FZ forum](https://forum.flipperzero.one/) for this kind of question
Your tag is a 7-byte UID tag and magic cards writing is not supported yet for 7-byte UID cards. FZ currently supports gen1a cards and they do not exist in 7-byte UID flavour