flipperdevices/flipperzero-firmware #2359: Pre-recorded "READ" files stopped working.

GR-noob, 2023-02-04:
### Describe the bug.

I was at (i think) 0.74.0 update and i updated to 0.76.0. I found that my flipper the sub files that i READ and saved stopped working but the READ RAW saved files still worked without problem. I re-record the 2 remotes in READ saved file again didn't work at READ RAW worked ok. I downgraded my flipper to 0.74.2 and started work again ok with the pre-recorded just RAED files started working again.

### Reproduction

just that pre-recorded just READ saved files didn't emulate anymore at the new update.

Skorpionm, 2023-02-05:
attach non-working files saved in read

skotopes, 2023-03-03:
Issue was fixed in #2443 please check and let us know if it still persist

GR-noob, 2023-03-14:
Hello. I updated my flipper to 0.79.1 and kinda again READ and saved stopped working. I re-READ the remote and this time works ok is just the old saved don't work anymore. Tomorrow i will test the file above ( commented on Feb 6 ) if it works for the other door or is just my door don't work. I will post again tomorrow.
The diference of the saved file of the old:
[image]
And new:
[image]

skotopes, 2023-03-15:
Can you provide us raw signal for analysis?

GR-noob, 2023-03-18:
i will upload 3 files. Is from the same door where i can provide without problem. OLD is the file who don't work anymore. NEW is the new READ save who works now and the OLD RAW is the RAW save from the old firmware who also works.

https://drive.google.com/file/d/1fH5002u3gCRRcuRSrjeoCa0_YlVNy49S/view?usp=sharing
https://drive.google.com/file/d/1QN1ykV_9IJBgtHFgjcTidmAOXbMJNSB4/view?usp=sharing
https://drive.google.com/file/d/1B9vONsHPLMojed463_JQpDdfPHiDbm3Q/view?usp=sharing

Ethanol0001, 2023-03-28:
@skotopes you think this might have been caused by the removal of the furi value mutex in the last update?

Ethanol0001, 2023-03-28:
wondering if the old files cant be recognized if its saving with the new system and is unable to read files still containing stuff that uses the furi value mutex

GR-noob, 2023-04-12:
Version 0.81.1 and still can't sent old files. I am the only one with this problem ?? I am so stupid i do something wrong ?? I am just curious if that cause problems with files i downloaded for different devises like some selling fans.

skotopes, 2023-04-13:
> Version 0.81.1 and still can't sent old files. I am the only one with this problem ?? I am so stupid i do something wrong ?? I am just curious if that cause problems with files i downloaded for different devises like some selling fans.

What exactly flipper tells you when you trying to send?

skotopes, 2023-04-13:
@Skorpionm regarding TE to Holtek_HT12X, is default value(TE is not set) same as old one?

GR-noob, 2023-04-13:
> > Version 0.81.1 and still can't sent old files. I am the only one with this problem ?? I am so stupid i do something wrong ?? I am just curious if that cause problems with files i downloaded for different devises like some selling fans.
> 
> What exactly flipper tells you when you trying to send?

I click SENT the led flashes as always but door don't open and nothing show at the screen except the 2 lines KEY and YEK. Do i downgrade again back to  0.74.2 to see if file still works ?? Because that was the last firmware it worked

Skorpionm, 2023-04-13:
after firmware 0.74.2, the CAME protocol timings were changed and a new Holtek protocol was added. The system was not correctly recognized in your old saved files, but with the old CAME protocol settings, your receiver could somehow still recognize the signal, now it simply does not recognize it, since the timings in CAME have been corrected to be more accurate. You just need to accept the data from the original remote control of your system again in the new firmware, save it and use it

Skorpionm, 2023-04-13:
[image]
all your files are playable and working

closed 2023-04-13 by skotopes