flipperdevices/flipperzero-firmware #2363: Air Mouse - Not opening

breadberry-prime, 2023-02-06:
### Describe the bug.

When trying to open Air Mouse it closes again.

### Reproduction

1. Switch on
2. Go to GPIO_Extra
3. Open Air Mouse

https://user-images.githubusercontent.com/75136042/216921391-bf205b6e-4070-4ff2-98a3-1e96f1a9796a.MOV

4. Dosen't open

G-haoyu, 2023-02-09:
Air Mouse is a community project and I suggest you to mention issue in the air mouse project.
Second, Air Mouse requires additional hardware support, and the author of the project clearly states in his README the type of hardware required (BMI160 inertial measurement unit).
If this hardware is not plugged in, the program will automatically exit, just like you show in your video.

skotopes, 2023-02-10:
Yep, this is clearly not a part of official firmware. Please report it to appropriate project.

closed 2023-02-10 by skotopes