flipperdevices/flipperzero-firmware #2385: MiZiP keys and card read support

LowSkillDeveloper, 2023-02-11:
### Description of the feature you're suggesting.

I have a "MiZiP" key for the coffee machine. As I understand it, they come in 2 types, NFC and RFID. And they contain information about the balance on the key. But the flipper can't read the balance from these keys:

[image]

My flipper and phone does not see the NFC, so I can assume that the key is RFID.
When reading RFID with Flipper Zero, it gives this:

[image]

> Filetype: Flipper RFID key
> Version: 1
> Key type: EM4100
> Data: 00 00 00 00 00

I also did "Read RAW RFID data" when the balance on the key was different. 88 coins, 82, 78, 66.
I don't know how and which program use to read RAW files, so I'll just attach them here:
https://github.com/LowSkillDeveloper/mizip-flipper-temp

Having a dump of the same key with different balances, I think it is possible to find a changing value

Also this is official site with information about MiZiP: https://newis.evocagroup.com/en/payment-solutions/mizip
Perhaps they also use specific NFC, so the flipper can't detect it?

augustozanellato, 2023-02-13:
Mizip stands for Mifare Zip, it’s a mifare mini tag. 
Lf-rfid ZIP keys are based on pcf7931 which flipper doesn’t support due to hw limitations iirc

skotopes, 2023-02-13:
@augustozanellato theoretically pcf7931 can be implemented on top of flipper hardware. At least 5577 is working fine. Question is how stable it will be.

augustozanellato, 2023-02-13:
> @augustozanellato theoretically pcf7931 can be implemented on top of flipper hardware. At least 5577 is working fine. Question is how stable it will be.

Oh, so read only would be feasible? Last time I asked about PCF transponder in discord I was told that it wouldn’t be possible due to hw limitations. 
cfr https://discord.com/channels/740930220399525928/954422698879098990/1040213253915156560

skotopes, 2023-02-13:
Yep, there are some challenges described by @DrZlo13 ;-)

augustozanellato, 2023-02-13:
> Yep, there are some challenges described by @DrZlo13 ;-)

But I guess you like challenges😂

more info on MiZIP btw: 
Key derivation from uid: https://github.com/RfidResearchGroup/proxmark3/blob/master/client/luascripts/hf_mf_uidkeycalc_mizip.lua
Data encoding: https://slides.com/siras/chibre/fullscreen

LowSkillDeveloper, 2024-06-22:
Are there any updates?

nocturn9x, 2025-04-17:
I have one of those and I can read it with my smartphone just fine. Maybe there's both NFC and RFID models?

LowSkillDeveloper, 2025-04-17:
> Maybe there's both NFC and RFID models?

Yes, as I understand there are 2 versions. But I wonder how the data saving works in the rfid version... Of course, there is an option that there is no data there, and everything is on the server in the database and a simple id binding. But as we can see, the id on my photo and other data are zero, but the keys works.

nocturn9x, 2025-04-17:
I'm not familiar with RFID devices, so I can't say. I've messed a bunch with MiZip NFC keys though, I should dig mine back up actually.

But yeah, if it's RFID then 🤷

grugnoymeme, 2026-11-10:
i implemented pcf7930/31/35 reading and writing.. it is stable... the only thing is not possible due to hw limitations is to interact with the pit 2 ones, bc those are not real pcf, but attiny406 emulating a pcf, and those required more power to be red..