flipperdevices/flipperzero-firmware #2386: Indala card not being read

Divide-By-0, 2023-02-12:
### Describe the bug.

I have a dual Indala 125 kHz ID and Mifare Classic 13.56 mHz card, and the Mifare classic reads fine on NFC. The Indala should read on the Extra Actions menu but nothing is detected.

### Reproduction

Switch to Indala mode
Try to read
It doesn't read

### Target

Indala card

### Logs

```Text
N/A
```

### Anything else?

I can read it on the Proxmark 3, it starts with 80000003...

umantouch69, 2023-02-12:
No entiendo

DrZlo13, 2023-02-12:
Yes, Indala224 is not supported by the LF-RFID protocol library. We need to wait for someone to implement it.

Divide-By-0, 2023-02-13:
I see, is it still possible to emulate/write if I know the ground truth code from being previously read?

Codeplaya, 2023-03-31:
It seems that Indala 64-bit and 224 bit cards require different demodulation. Looking at [Proxmark 3 Indala options](https://github.com/Proxmark/proxmark3/blob/555fa197730c061bbf0ab01334e99bc47fb3dc06/client/cmdlfindala.c), there are are besides PSK1 and rf/32 also 64 and 224 bits Indala cards that have a whole different [demodulation](https://github.com/Proxmark/proxmark3/search?q=indala64decode).

closed 2026-10-07 by turbospok