flipperdevices/flipperzero-firmware #2404: Support for sub-GHz RADIAN protocol

adrianlzt, 2023-02-16:
### Description of the feature you're suggesting.

There are some water/gas smart meters (EverBlu Cyble) which use the RADIAN protocol (frequency 433MHz) to transmit their data. [This project](https://github.com/neutrinus/everblu-meters) uses a Raspberry Pi + CC1101 to read the data.

I was thinking it could be useful to have an application like _Weather Station_ to read these kinds of meters.

Initial thoughts about how complicated the implementation could be?

Keep in mind that bidirectional communication is needed, as the meter will only send data upon request.

Skorpionm, 2023-02-19:
there is no complexity, there is a Cli application "subghz chat", it implements 2-way communication in batch mode, you can make such a Fap

skotopes, 2023-02-26:
Yep, this thing is more complex than WeatherStation. Clearly deserves its own application.

closed 2023-02-26 by skotopes