flipperdevices/flipperzero-firmware #2413: Flipper not reading a sector in spite of having all keys

Alejandro12120, 2023-02-18:
### Describe the bug.

This is a very weird bug, let me explain. I have a gym (Mifare Classic 1k) card which after scanning it with my Flipper Zero it returns 32/32 keys found and 15/16 sectors. 
Therefore I checked the access bits for that sector.   
[image] 
 It looks like you can only read the sector using key B but in my dump Flipper knowns that key. Also if I emulate dump the card with my proxmark3 and emulates it, Flipper is able to read all the sectors, but if I load the dump to a magic gen1a tag using proxmark, Flipper is only able to read 15/16 sectors.

## Dumps
[dumps.zip](https://github.com/flipperdevices/flipperzero-firmware/files/10773816/dumps.zip)

### Reproduction

1. Add `9438DBD47284` to dictionary
2. Emulate `gym_proxmark.bin` using a Proxmark
3. Read it using Flipper
4. You should see that 16/16 sectors are read
5. Load `gym_proxmark.bin` to a gen1a tag using Proxmark
6. Read it using Flipper
7. You should see that 15/16 sectors are read 

https://user-images.githubusercontent.com/45500329/219854094-3a100826-a176-47b8-9d57-a079d5987835.mp4

UrSuLa360, 2023-02-20:
Hello,
same bug here. Just to provide some light.
Mifare classic 1k tag with 5 custom B keys known and already added in user dictionary.

version 0.76 and up

[image]

and version 0.75 and down
[image]

Thank you for your work! You all rock!

UrSuLa360, 2023-02-20:
`Firmware version: 0.77.1 0.77.1 (a38a62fe built on 14-02-2023)

>: log
Press CTRL+C to stop...
1169582 [I][LoaderSrv] Starting: NFC
1169591 [I][AnimationManager] Unload animation 'L2_Hacking_pc_128x64'
1170223 [D][DolphinState] icounter 863, butthurt 0
1171190 [I][NfcWorker] Mifare Classic detected
1171194 [I][NfcWorker] Trying to read a supported card ...
1171200 [D][Plant] Verifying sector 8
1171259 [D][FuriHalNfc] Timeout during data exchange
1171263 [D][Troika] Verifying sector 11
1171315 [D][FuriHalNfc] Timeout during data exchange
1171319 [I][NfcWorker] Search for key cache ...
1171444 [I][NfcWorker] Load keys cache success. Start reading
1171447 [D][MfClassic] Try to read blocks with key A
1171524 [D][MfClassic] Read 4 blocks out of 4
1171528 [D][MfClassic] Try to read blocks with key A
1171598 [D][MfClassic] Read 4 blocks out of 4
1171602 [D][MfClassic] Try to read blocks with key A
1171672 [D][MfClassic] Read 4 blocks out of 4
1171676 [D][MfClassic] Try to read blocks with key A
1171746 [D][MfClassic] Read 4 blocks out of 4
1171750 [D][MfClassic] Try to read blocks with key A
1171820 [D][MfClassic] Read 4 blocks out of 4
1171824 [D][MfClassic] Try to read blocks with key A
1171899 [D][MfClassic] Read 4 blocks out of 4
1171903 [D][MfClassic] Try to read blocks with key A
1171973 [D][MfClassic] Read 4 blocks out of 4
1171977 [D][MfClassic] Try to read blocks with key A
1172047 [D][MfClassic] Read 4 blocks out of 4
1172051 [D][MfClassic] Try to read blocks with key A
1172121 [D][MfClassic] Read 4 blocks out of 4
1172125 [D][MfClassic] Try to read blocks with key A
1172200 [D][MfClassic] Read 4 blocks out of 4
1172204 [D][MfClassic] Try to read blocks with key A
1172274 [D][MfClassic] Read 4 blocks out of 4
1172278 [D][MfClassic] Try to read blocks with key A
1172438 [D][FuriHalNfc] Timeout during data exchange
1172542 [D][FuriHalNfc] Timeout during data exchange
1172598 [D][MfClassic] Read 2 blocks out of 4
1172601 [D][MfClassic] Try to read blocks with key B
1172804 [D][MfClassic] Try to read blocks with key A
1172910 [D][FuriHalNfc] Timeout during data exchange
1173014 [D][FuriHalNfc] Timeout during data exchange
1173118 [D][FuriHalNfc] Timeout during data exchange
1173178 [D][MfClassic] Read 1 blocks out of 4
1173181 [D][MfClassic] Try to read blocks with key B
1173385 [D][MfClassic] Try to read blocks with key A
1173498 [D][FuriHalNfc] Timeout during data exchange
1173605 [D][FuriHalNfc] Timeout during data exchange
1173709 [D][FuriHalNfc] Timeout during data exchange
1173765 [D][MfClassic] Read 1 blocks out of 4
1173768 [D][MfClassic] Try to read blocks with key B
1173971 [D][MfClassic] Try to read blocks with key A
1174077 [D][FuriHalNfc] Timeout during data exchange
1174181 [D][FuriHalNfc] Timeout during data exchange
1174288 [D][FuriHalNfc] Timeout during data exchange
1174344 [D][MfClassic] Read 1 blocks out of 4
1174347 [D][MfClassic] Try to read blocks with key B
1174550 [D][MfClassic] Try to read blocks with key A
1174656 [D][FuriHalNfc] Timeout during data exchange
1174760 [D][FuriHalNfc] Timeout during data exchange
1174821 [D][MfClassic] Read 2 blocks out of 4
1174824 [D][MfClassic] Try to read blocks with key B
1175026 [D][MfClassic] Read 11 sectors and 32 keys
1175028 [I][NfcWorker] Read 11 sectors out of 16 total
1175050 [I][MfClassicDict] Loaded dictionary with 5 keys
1175057 [D][NfcWorker] Start Dictionary attack, Key Count 5
1175063 [I][NfcWorker] Sector 0
[... 67 lines cut ...]
1178502 [D][MfClassic] Read 1 blocks out of 4
1178506 [D][MfClassic] Try to read blocks with key B
1178710 [I][NfcWorker] Sector 13
1178765 [D][NfcWorker] Try to auth to sector 13 with key ffffffffffff
1178768 [D][MfClassic] Try to read blocks with key A
1178874 [D][FuriHalNfc] Timeout during data exchange
1178981 [D][FuriHalNfc] Timeout during data exchange
1179035 [D][MfClassic] Read 1 blocks out of 4
1179037 [D][MfClassic] Try to read blocks with key B
1179240 [I][NfcWorker] Sector 14
1179295 [D][NfcWorker] Try to auth to sector 14 with key ffffffffffff
1179298 [D][MfClassic] Try to read blocks with key A
1179404 [D][FuriHalNfc] Timeout during data exchange
1179508 [D][FuriHalNfc] Timeout during data exchange
1179572 [D][MfClassic] Read 1 blocks out of 4
1179576 [D][MfClassic] Try to read blocks with key B
1179780 [I][NfcWorker] Sector 15
1179835 [D][NfcWorker] Try to auth to sector 15 with key ffffffffffff
1179838 [D][MfClassic] Try to read blocks with key A
1179944 [D][FuriHalNfc] Timeout during data exchange
1179998 [D][MfClassic] Read 2 blocks out of 4
1180000 [D][MfClassic] Try to read blocks with key B
1180412 [D][DolphinState] icounter 866, butthurt 0`

Log added

UrSuLa360, 2023-02-21:
`Firmware version: dev unknown (663eb6cd built on 21-02-2023)

>: log
Press CTRL+C to stop...
77246 [I][FuriHalNfc] Deinit OK
77376 [I][FuriHalNfc] Init OK
77380 [D][DolphinState] icounter 871, butthurt 0
78345 [I][NfcWorker] Mifare Classic detected
78349 [I][NfcWorker] Trying to read a supported card ...
78355 [D][Plant] Verifying sector 8
78414 [D][FuriHalNfc] Timeout during data exchange
78417 [D][Troika] Verifying sector 11
78470 [D][FuriHalNfc] Timeout during data exchange
78473 [I][NfcWorker] Search for key cache ...
78587 [I][NfcWorker] Load keys cache success. Start reading
78589 [D][MfClassic] Try to read blocks with key A
78664 [D][MfClassic] Read 4 blocks out of 4
78667 [D][MfClassic] Try to read blocks with key A
78739 [D][MfClassic] Read 4 blocks out of 4
78742 [D][MfClassic] Try to read blocks with key A
78814 [D][MfClassic] Read 4 blocks out of 4
78817 [D][MfClassic] Try to read blocks with key A
78895 [D][MfClassic] Read 4 blocks out of 4
78898 [D][MfClassic] Try to read blocks with key A
78974 [D][MfClassic] Read 4 blocks out of 4
78977 [D][MfClassic] Try to read blocks with key A
79049 [D][MfClassic] Read 4 blocks out of 4
79052 [D][MfClassic] Try to read blocks with key A
79124 [D][MfClassic] Read 4 blocks out of 4
79127 [D][MfClassic] Try to read blocks with key A
79199 [D][MfClassic] Read 4 blocks out of 4
79202 [D][MfClassic] Try to read blocks with key A
79274 [D][MfClassic] Read 4 blocks out of 4
79277 [D][MfClassic] Try to read blocks with key A
79354 [D][MfClassic] Read 4 blocks out of 4
79357 [D][MfClassic] Try to read blocks with key A
79429 [D][MfClassic] Read 4 blocks out of 4
79432 [D][MfClassic] Try to read blocks with key A
79594 [D][FuriHalNfc] Timeout during data exchange
79701 [D][FuriHalNfc] Timeout during data exchange
79757 [D][MfClassic] Read 2 blocks out of 4
79760 [D][MfClassic] Try to read blocks with key B
79963 [D][MfClassic] Try to read blocks with key A
80069 [D][FuriHalNfc] Timeout during data exchange
80173 [D][FuriHalNfc] Timeout during data exchange
80277 [D][FuriHalNfc] Timeout during data exchange
80337 [D][MfClassic] Read 1 blocks out of 4
80340 [D][MfClassic] Try to read blocks with key B
80544 [D][MfClassic] Try to read blocks with key A
80648 [D][FuriHalNfc] Timeout during data exchange
80755 [D][FuriHalNfc] Timeout during data exchange
80859 [D][FuriHalNfc] Timeout during data exchange
80922 [D][MfClassic] Read 1 blocks out of 4
80925 [D][MfClassic] Try to read blocks with key B
81129 [D][MfClassic] Try to read blocks with key A
81235 [D][FuriHalNfc] Timeout during data exchange
81342 [D][FuriHalNfc] Timeout during data exchange
81446 [D][FuriHalNfc] Timeout during data exchange
81502 [D][MfClassic] Read 1 blocks out of 4
81505 [D][MfClassic] Try to read blocks with key B
81708 [D][MfClassic] Try to read blocks with key A
81814 [D][FuriHalNfc] Timeout during data exchange
81918 [D][FuriHalNfc] Timeout during data exchange
81990 [D][MfClassic] Read 2 blocks out of 4
81993 [D][MfClassic] Try to read blocks with key B
82195 [D][MfClassic] Read 11 sectors and 32 keys
82197 [I][NfcWorker] Read 11 sectors out of 16 total
82212 [I][MfClassicDict] Loaded dictionary with 5 keys
82214 [I][FuriHalNfc] Deinit OK
82339 [I][FuriHalNfc] Init OK
[... 79 lines cut ...]
86480 [D][MfClassic] Try to read blocks with key B
86683 [I][NfcWorker] Sector 14
86737 [D][NfcWorker] Try to auth to sector 14 with key ffffffffffff
86740 [D][MfClassic] Try to read blocks with key A
86846 [D][FuriHalNfc] Timeout during data exchange
86950 [D][FuriHalNfc] Timeout during data exchange
87007 [D][MfClassic] Read 1 blocks out of 4
87011 [D][MfClassic] Try to read blocks with key B
87215 [I][NfcWorker] Sector 15
87270 [D][NfcWorker] Try to auth to sector 15 with key ffffffffffff
87273 [D][MfClassic] Try to read blocks with key A
87379 [D][FuriHalNfc] Timeout during data exchange
87433 [D][MfClassic] Read 2 blocks out of 4
87435 [D][MfClassic] Try to read blocks with key B
87847 [D][DolphinState] icounter 874, butthurt 0`

New log with latest dev build

btw those log lines says :
84987 [I][MfClassicDict] Loaded dictionary with **1242** keys
84993 [I][FuriHalNfc] Deinit OK
85124 [I][FuriHalNfc] Init OK
85128 [D][NfcWorker] Start Dictionary attack, Key Count **1242**

And that number is only the default dictionary key count. I have 5 more keys in my user dict. Maybe dictionary attack is only using default dict keys.

dogtopus, 2023-02-24:
This seems to happen when the tag is using B key authentication.

dogtopus, 2023-02-26:
Commit that introduced this: https://github.com/flipperdevices/flipperzero-firmware/commit/8fc834090d289ad75de8261ad69ec310a39b5834

Looks like for all cards that I tested, the first keyA auth fail will cause the card to stop responding to subsequent attempts at authentication. Therefore a reset/`furi_hal_nfc_sleep()` call will still be needed here (with the exception that when we know for the first time that we should use keyB instead of keyA, we could directly use keyB for auth and skip the reset/`furi_hal_nfc_sleep()` call).

skotopes, 2023-03-07:
fixed in #2437

Please reopen if you still have this issue.

closed 2023-03-07 by skotopes

UrSuLa360, 2023-03-07:
Working fine! Thank you all!!!

Petieb1, 2023-03-12:
Just switched to dev update channel, running 4bd3dca1. Still having this issue, not sure if I am doing something else wrong here. Any advice is appreciated!

dogtopus, 2023-03-13:
> Just switched to dev update channel, running [4bd3dca](https://github.com/flipperdevices/flipperzero-firmware/commit/4bd3dca16f79d66b513d6505abe30a937a9fc7e5). Still having this issue, not sure if I am doing something else wrong here. Any advice is appreciated!

Make sure your card has the right access condition. Some access conditions will disable all access to the sector.

Petieb1, 2023-03-14:
> Make sure your card has the right access condition. Some access conditions will disable all access to the sector.

Sorry, I know this probably isn't the best forum for this, but it looks like the card has the same access bits for all sectors. 
Heres a selection of the read from the proxmark:
> [=]    1 |   4 | 3C 00 47 00 00 00 00 00 00 00 00 C1 00 00 00 45 | 
> [=]       |   5 | 3C 00 47 00 00 00 00 00 00 00 00 C1 00 00 00 45 | 
> [=]       |   6 | 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 | 
> [=]       |   7 | 2A 2C 13 CC 24 2A FF 07 80 69 FF FF FF FF FF FF | 
> [=]    2 |   8 | 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 | 
> [=]       |   9 | 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 | 
> [=]       |  10 | 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 | 
> [=]       |  11 | FF FF FF FF FF FF FF 07 80 69 FF FF FF FF FF FF | 
> [=]    3 |  12 | 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 | 
> [=]       |  13 | 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 | 
> [=]       |  14 | 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 | 
> [=]       |  15 | 21 22 52 22 19 DF FF 07 80 69 FF FF FF FF FF FF | 
> [=]    4 |  16 | 42 40 9C BA DB 20 00 00 AA 64 14 92 02 40 00 F6 | 
> [=]       |  17 | AA 64 44 F4 22 40 00 A8 AA 65 14 68 22 40 00 ED | 
> [=]       |  18 | 2A 65 14 95 01 C8 7B 7C 2A 65 14 95 01 C8 75 76 | 
> [=]       |  19 | 21 22 52 22 19 DF FF 07 80 69 FF FF FF FF FF FF | 

Here's what the read of the same sectors looks like from my FlipperZero:
> [=]    1 |   4 | ?? ?? ?? ?? ?? ?? ?? ?? ?? ?? ?? ?? ?? ?? ?? ?? | 
> [=]       |   5 | ?? ?? ?? ?? ?? ?? ?? ?? ?? ?? ?? ?? ?? ?? ?? ?? | 
> [=]       |   6 | ?? ?? ?? ?? ?? ?? ?? ?? ?? ?? ?? ?? ?? ?? ?? ?? | 
> [=]       |   7 | 00 00 00 00 00 00  ?? ?? ?? ?? 00 00 00 00 00 00 | 
> [=]    2 |   8 | ?? ?? ?? ?? ?? ?? ?? ?? ?? ?? ?? ?? ?? ?? ?? ?? |  
> [=]       |   9 | ?? ?? ?? ?? ?? ?? ?? ?? ?? ?? ?? ?? ?? ?? ?? ?? | 
> [=]       |  10 | ?? ?? ?? ?? ?? ?? ?? ?? ?? ?? ?? ?? ?? ?? ?? ?? | 
> [=]       |  11 | 00 00 00 00 00 00  ?? ?? ?? ?? 00 00 00 00 00 00 | 
> [=]    3 |  12 | 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 | 
> [=]       |  13 | 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 | 
> [=]       |  14 | 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 | 
> [=]       |  15 | 21 22 52 22 19 DF FF 07 80 69 FF FF FF FF FF FF | 
> [=]    4 |  16 | 42 40 9C BA DB 20 00 00 AA 64 14 92 02 40 00 F6 | 
> [=]       |  17 | AA 64 44 F4 22 40 00 A8 AA 65 14 68 22 40 00 ED | 
> [=]       |  18 | 2A 65 14 95 01 C8 7B 7C 2A 65 14 95 01 C8 75 76 | 
> [=]       |  19 | 21 22 52 22 19 DF FF 07 80 69 FF FF FF FF FF FF | 

Flipper zero read every other sector including 0 just fine and it looks like access bits are all set to 0xff0780 transport config. Is there a reason that access condition is okay on some sectors and not on others? Again sorry if this isn't the best place  I'm on the discord @RadioFox if that is better!

dogtopus, 2023-03-14:
Your card has the access condition `ff0780`, which is the default "open" mode that uses keyA.

It does look like something is wrong though. The second sector uses a default key but it still can't be read. Maybe a card compatibility issue or bad coil?

Petieb1, 2023-03-14:
This is a fob that I use regularly so I know that the tag itself is functional. I was able to dump the data to a magic tag via pm3 which is also functional on all of the actual readers but has the same issue when read by the flipper. I even tried emulating (-1k) the tag info with the proxmark itself and reading that with the flipper, but again had the same issue. I also confirmed that all of these keys are correct in my user dictionary.

Petieb1, 2023-03-16:
Updated to 25FD3C34 and it is now working just fine!

skotopes, 2023-03-16:
Awesome ;-) Thanks to @Astrrra

PR #2437 (2023-03-07): NFC: Support reading Mifare Classic key B from sector trailer, reading sector with B key where A key can't read block,  Nfc Magic app not using NFC folder by default (in file select)
# What's new

- NFC: Support reading Mifare Classic key B from sector trailer and reusing it for other sectors

- NFC: Fix reading sector with B key where A key can't read block (fixes #2413)

- NFC: Fix Nfc Magic app not using NFC folder by default (in file select)

Notes: 
I modified `mf_classic_auth_attempt` because it isn't used anywhere and it was easy to do what I needed (and didn't break anything, but actua [... cut]
commit eb5dae1c (2023-03-07): NFC: Support reading Mifare Classic key B from sector trailer, reading sector with B key where A key can't read block,  Nfc Magic app not using NFC folder by default (in file select) (#2437)

* NFC: Support reading Mifare Classic key B from sector trailer and reusing it for other sectors
* NFC: Fix my pointer typo
* NFC: Fix reading sector with B key where A key can't read block (fixes #2413) and fix Nfc Magic app not using NFC folder by default (in file select)
* NFC: Fix strange bug