flipperdevices/flipperzero-firmware #2422: Problems with usb keyboard

Landon-Kuehner, 2023-02-23:
### Describe the bug.

When pressing a key while using the usb keyboard while not plugged in will cause as far as I can tell a crash. I've tried everything I can think of without using my pc including unplugging the sd card. Oddly plugging and unplugging the sd card flashes the the leds suggesting that the firmware hasn't crashed. Also the backlight will automatically power off but not back on.

### Reproduction

1. Have the latest flipper version
2.  Run the usb keyboard
3. Click a button (I clicked f5 but other buttons might work)
4. Crash your flipper

### Target

Usb keyboard

### Anything else?

My flipper has been down for around 15 minutes without snapping out of it.

Landon-Kuehner, 2023-02-23:
Plugging it into an active device for any amount of time seems to fix it and unfreeze it

skotopes, 2023-02-26:
@Landon-Kuehner there were multiple fixes and improvement added recently. Please reopen issue if problem still persists in latest dev.

closed 2023-02-26 by skotopes