flipperdevices/flipperzero-firmware #2425: Badbutton detection for Somfy Telis remote

costimasca, 2023-02-24:
### Describe the bug.

No matter what button I press on my Somfy Telis remote, Flipper will always detect the btn to be 0x0F.

### Reproduction

1. Go to SubGHZ
2. Go to Read
3. Set frequency to 433.42
4. Press a button on the Somfy Telis remote
5. Press another button on the Somfy Telis remote
6. Check the 2 decoded records - they will have the same code for the button

Skorpionm, 2023-02-26:
No, it won't, the decoder was written from the original remote control, what is your remote control?  and what is the firmware

skotopes, 2023-02-26:
@costimasca also can you provide recording of each button in RAW format?

costimasca, 2023-03-02:
Firmware used: latest = 0.78.1

I've attached 2 buttons: up and down. 

More info:
When using normal Read, the counter and the key do increase on each button press. 
Still, no matter what button I press, the Btn value is 0x0F.

I've also attached a picture of the remote. Not sure if it's original or just Somfy compatible??

[Archive.zip](https://github.com/flipperdevices/flipperzero-firmware/files/10870627/Archive.zip)
[image]

Skorpionm, 2023-03-02:
[image]

here is the original., I will look at your entries

skotopes, 2023-06-22:
Please reopen if your issue is still there

closed 2023-06-22 by skotopes

trueVinton, 2024-05-17:
Hi @skotopes, I am currently facing exactly the same issue described in this thread. Can we reopen it, please?

Additional context: I tried using "Add Manually" > Somfy Telis. In this case, I can see that pressing different buttons on Flipper sends different Btn codes. However, when I edit the previously created .sub file to adjust the remote SN and run it once again, it seems to break and only sends the Btn:0x0F value for any key pressed on the Flipper.

skotopes, 2024-05-17:
@Skorpionm wanna make 2nd take?