flipperdevices/flipperzero-firmware #2433: Unknown RFID (125 kHz) protocol support request

BinkyTheDoormat, 2023-02-26:
### Description of the feature you're suggesting.

I've recently found an unidentified 125kHz RFID access card that looks like T5577 (and has some advertisement text on it yet nothing useful) that cannot be read both with Flipper and Proxmark. I was able to capture its contents in raw: 
[Sus_Card-raw.zip](https://github.com/flipperdevices/flipperzero-firmware/files/10833725/Sus_Card-raw.zip)

Could you please help me identify what this is (in case raw file has something useful on it) and tell if this protocol can be implemented in Flipper firmware?

skotopes, 2023-02-26:
Are you sure that it is LF RFID? Also is it working properly?

BinkyTheDoormat, 2023-02-26:
Almost 100% sure after I've investigated its guts with a flashlight. It works properly on an access control system it's designed for

skotopes, 2023-02-27:
Also try to read it with NFC or Picopass app

BinkyTheDoormat, 2023-03-03:
Tried already with no success

skotopes, 2023-06-22:
ASK is empty, PSK doesn't contains valid data. Most likely it's not RFID. Try it with proxmark

closed 2024-07-07 by skotopes