flipperdevices/flipperzero-firmware #2442: gipo usb-uart bridge fails to work

Ethanol0001, 2023-02-27:
### Describe the bug.

while trying to connect to a computer to your flipper using the gipo usb-utart bridge connection is able to read data but the computer is unable to reconize the flipper as a device this way 

### Reproduction

use a gipo to usb conversion tool to connect the flipper to, use male to female jumper cables and connect it appropriately to the converter, set the config on the flipper to match that of the device you are attempting to connect the flipper to.

### Target

Laptop USB via GIPO usb-utart connection

### Anything else?

The computer is able to detect something plugging into it but isn't registering it as a device. (will add debug logs here later when I get home and reatempt this

skotopes, 2023-03-13:
We still need logs. From your host operation system and flipper (at least till the moment of mode switch)

skotopes, 2023-03-13:
Also versions of flipper firmware and host OS

Ethanol0001, 2023-03-28:
not sure if there would even be any logs due to it only sending 1 byte of data to the computer when it detects power

skotopes, 2023-03-29:
which port settings are you using?

closed 2023-06-22 by skotopes