flipperdevices/flipperzero-firmware #2445: Custom SubGhz protocols with logics

alvarotorijano, 2023-02-27:
### Description of the feature you're suggesting.

Hi, I would like to be able to add protocols that involve some logic. For example protocols with a specific bitStream initializer and those which involve some logics with the dip switches.
I suggest to use the following interface

`void dipSwitchToProtocol(unsigned char dipSw, unsigned char bitStream[PROTOCOL_LENGTH])`
`int protocolToDipSwitch(unsigned char byte[PROTOCOL_LENGTH])`
`uint8_t idetifyProtocol ( uint8_t * bitStream, int size)`

Skorpionm, 2023-03-06:
it does not make sense. since even consoles with DIP switches include different sets of bit lengths, different types of bit encoding, preambles, durations, etc. for custom protocols there is BinRaw, you can emit almost any protocols in it

alvarotorijano, 2023-03-06:
Sending protocols as raw does not allow to identify or decode received codes and the way for defining new protocols in flipper is too simple. As user I want to be able to add complex protocols further than so simple way used in the SDK. And this is not just a deal, I've managed to decode a protocol that involves some logic and I cannot include in the flipper for it to detect and decode de dipswitches. So please reconsider this request as it is not senseless.

Skorpionm, 2023-03-06:
if you have protocol encoding logic, you can, just, normally add it like the rest

closed 2023-03-13 by skotopes