flipperdevices/flipperzero-firmware #2449: Support for Fairpointe Data RFID cards and readers

g1eb, 2023-03-01:
### Description of the feature you're suggesting.

Looks like the flipper is not able to read [Fairpointe Data cards](https://www.farpointedata.com/downloads/datasheets/DC1_TDS.pdf) (yet) .. is this correct?

The reader looks like this pyramid here: [https://www.farpointedata.com/downloads/datasheets/EM_TDS.pdf](https://www.farpointedata.com/downloads/datasheets/EM_TDS.pdf)

There is an option to add Fairpointe Pyramid hex manually but I'm not sure how to generate that, if anyone here understands this technology could weigh in I'd really appreciate the help! Here's some documentation I found about the protocol that they use: [https://www.farpointedata.com/downloads/pr/Understanding_Wiegand.pdf](https://www.farpointedata.com/downloads/pr/Understanding_Wiegand.pdf)

dogtopus, 2023-03-02:
Isn't it just a Mifare Classic 1k?

skotopes, 2023-03-13:
You've mention a bunch of different technologies, which specific card do you have?
What exactly RFID/NFC app tells you?

g1eb, 2023-03-13:
Hi! It's a Fairpointe card, just like the one shown [here](https://www.farpointedata.com/downloads/datasheets/DC1_TDS.pdf) and the reader looks like [this pyramid here ](https://i.imgur.com/jxTirTa.jpg)

The RFID read keeps hanging on the the ASK/PSK screen and does not read anything.. [see picture](https://i.imgur.com/o8RrWQ3.jpeg)

> What exactly RFID/NFC app tells you?

Sorry could you be more specific? Do you mean to use the NFC read and record the output of that? I can try this next time that i'm on site.

dogtopus, 2023-03-13:
> Hi! It's a Fairpointe card, just like the one shown [here](https://www.farpointedata.com/downloads/datasheets/DC1_TDS.pdf) and the reader looks like [this pyramid here [...]

This is not helpful as previously suggested by @skotopes. Your documentation mentions Mifare Classic 1k but you are scanning with 125kHz RFID?

closed 2023-06-22 by skotopes