flipperdevices/flipperzero-firmware #2453: Locking the Flipper does not turn off right away the display

ErSeraph, 2023-03-02:
### Describe the bug.

Holding UP let you lock your Flipper (with PIN if already set), pressing a button displays a message, and after 1 second the display turn off every time until the unlock code or back 3 times are used.

Personally I use the no-PIN lock system every time, but I noticed that the display will not turn off the first time you lock the flipper, you have to wait <BACKLIGHT TIME SETTING> seconds. This confuse me every time because I don't know if I locked it correctly, I know that there is an icon upper left but I wanted to warn about this.

### Reproduction

1. Lock your Flipper
2. Display does not turn off right away

ErSeraph, 2023-03-03:
@andr0423 I think you referenced the wrong issue

Ethanol0001, 2023-03-03:
are you talking about the animation that plays? Because the animation plays first before it locks. Also mind providing what version you are using such as release 0.78.1 or the release canadate?

ErSeraph, 2023-03-03:
Sorry, I didn't thought that Im using a CF, but it's based on version 0.78.1. Im talking about that after you lock the flipper it will not turn off the screen, you have to wait 30 seconds or what you set in the "LCD and Notification" setting. Every time you press a button it say "To unlock press BACK BACK BACK" and turn off the display after one second, but not the first time. It's not a real problem but I have to check every time if I locked it correctly before putting in my pocket.

andr0423, 2023-03-05:
> @andr0423 I think you referenced the wrong issue

Oh,  sorry for mistake.

polarikus, 2023-03-09:
Есть видео? 
+ Приложить инфу о прошивке (Долго удерживать вниз, приложить скрин ю фото с инфой). 
Я не могу воспроизвести то, что описано на своём флиппере.

ErSeraph, 2023-03-09:
I want to apologize to everyone, this issue is related to a custom firmware Im using and not to the official one, sorry for the mistake and wasting time.