flipperdevices/flipperzero-firmware #2460: SubGHz: Error in protocol parameters description

agilly, 2023-03-05:
### Describe the bug.

I wanted to clone my ceiling fan remote (FCC ID A25-43). Replaying the RAW capture works, but once decoded I cannot replay the .sub file. I get a "Error in protocol parameters description".

This was run on the latest release and the latest dev firmware as of today, the error happens on both.

### Reproduction

Try to read the following file:
```
Filetype: Flipper SubGhz Key File
Version: 1
Frequency: 303875000
Preset: FuriHalSubGhzPresetOok650Async
Protocol: GateTX
Bit: 32
Key: 00 00 00 00 41 F0 FD 7E
```

The corresponding RAW works:
```
Filetype: Flipper SubGhz RAW File
Version: 1
Frequency: 303875000
Preset: FuriHalSubGhzPresetOok650Async
Protocol: RAW
RAW_Data: 331 -68 1057 -98 723 -66 221611 -68 331 -66 569 -98 987 -134 397 -98 205547 -66 43471 -100 34869 -102 5419 -66 14241 -100 81581 -98 263 -64 459 -64 231 -64 327 -132 331 -68 199 -68 21291 -68 14835 -66 42273 -66 3913 -64 15621 -66 2375 -98 12209 -68 959 -66 12107 -100 45709 -98 21243 -66 31813 -68 1265 -66 31779 -66 12377 -66 1565 -66 9953 -66 3681 -66 3291 -66 1849 -66 2567 -98 1753 -66 30943 -100 20983 -66 2699 -66 1655 -68 565 -66 535 -66 6103 -132 231 -66 12481 -100 233 -298 3013 -100 667 -132 429 -496 529 -330 395 -166 165 -168 5013 -132 3723 -100 733 -100 5421 -132 1911 -68 433 -64 20737 -66 11435 -37992 65 -4672 763 -266 165 -98 3075 -68 30261 -66 4975 -100 5429 -66 4143 -68 4685 -66 5733 -66 1911 -66 3573 -100 1997 -66 4687 -98 657 -66 1283 -66 11191 -98 2133 -100 12077 -66 2341 -100 11855 -100 3447 -100 2393 -66 1293 -98 833 -66 2451 -66 1191 -66 6891 -98 897 -66 5641 -66 1189 -68 4105 -66 4299 -100 1087 -66 1307 -68 797 -66 829 -66 4527 -794 385 -408 811 -384 805 -382 807 -386 801 -382 807 -786 415 -770 413 -784 415 -770 413 -782 415 -378 807 -414 769 -386 807 -416 803 -774 381 -808 417 -768 417 -774 421 -780 381 -820 389 -386 807 -778 409 -378 833 -774 403 -800 379 -806 411 -768 411 -778 415 -770 413 -420 777 -13994 417 -458 765 -788 395 -418 791 -388 803 -394 795 -404 783 -412 803 -776 407 -788 395 -794 405 -790 395 -790 405 -380 799 -408 803 -386 803 -414 769 -808 381 -810 383 -804 383 -806 383 -802 385 -806 421 -384 781 -812 381 -392 799 -808 403 -790 395 -790 403 -806 387 -804 377 -806 385 -410 777 -14014 413 -452 783 -782 391 -416 789 -388 805 -396 791 -406 785 -396 795 -804 403 -790 397 -788 405 -790 395 -792 403 -398 787 -406 787 -396 795 -404 787 -820 357 -832 377 -788 391 -830 375 -788 393 -796 407 -396 793 -800 399 -380 797 -804 403 -802 379 -806 381 -818 387 -802 379 -818 357 -436 779 -14020 413 -416 811 -780 387 -418 789 -388 801 -396 795 -404 787 -394 795 -804 403 -786 417 -788 385 -786 415 -770 417 -386 815 -384 801 -398 787 -404 787 -784 411 -804 379 -788 391 -798 411 -786 391 -798 409 -396 791 -804 399 -378 795 -806 401 -806 387 -772 409 -804 385 -808 379 -804 383 -412 811 -13994 413 -418 785 -826 363 -416 797 -390 801 -380 799 -408 783 -424 767 -808 407 -788 393 -796 405 -790 393 -794 407 -396 789 -406 785 -396 795 -406 787 -788 391 -796 409 -786 409 -806 379 -816 379 -806
RAW_Data: 379 -394 795 -804 405 -396 789 -798 403 -786 379 -798 407 -806 385 -806 377 -802 381 -410 803 -14014 413 -418 783 -826 363 -416 797 -390 797 -418 789 -386 803 -416 767 -786 395 -820 385 -782 415 -788 385 -812 381 -418 781 -384 805 -420 781 -382 805 -786 419 -782 383 -810 411 -772 383 -812 415 -782 385 -410 779 -810 383 -418 779 -812 387 -802 375 -820 401 -784 385 -816 383 -802 377 -402 815 -13996 411 -450 783 -784 389 -408 781 -428 771 -392 797 -410 787 -392 797 -806 403 -788 395 -794 407 -790 393 -794 405 -398 789 -404 789 -396 793 -404 787 -786 391 -800 409 -790 391 -796 407 -802 379 -804 411 -380 803 -774 409 -410 769 -804 405 -802 377 -808 411 -772 383 -810 413 -772 413 -382 805 -14020 413 -418 781 -790 395 -410 793 -406 787 -394 795 -406 783 -412 805 -776 407 -790 395 -794 407 -788 395 -790 407 -398 789 -402 805 -388 801 -380 817 -782 385 -808 415 -782 385 -806 381 -786 391 -798 411 -396 793 -804 403 -366 819 -800 401 -786 383 -818 385 -788 419 -780 383 -808 385 -388 805 -146582 2583 -996 131 -732 263 -166 165 -134 97 -496 229 -132 99 -132 100471 -132 3203 -66 1631 -100 3403 -98 4927 -100 1163 -100 3699 -66 4881 -98 329 -64 1287 -98 633 -134 4829 -66 2295 -66 2899 -66 263 -100 1115 -196 531 -66 11613 -66 1925 -66 4825 -100 20903 -100 10335 -66 425 -68 2647 -98 363 -68 5607 -100 267 -68 299 -66 301 -100 21231 -66 14181 -66 1513 -66 559 -66 2547 -98 5425 -68 2011 -66 2431 -66 4731 -66 1255 -66 365 -66 1795 -164 3389 -66 1431 -100 499 -66 4313 -100 3383 -66 701 -98 4547 -98 4639 -132 1417 -98 3495 -66 21713 -98 785 -66 663 -66 5667 -166 10613 -68 1909 -66 233 -66 13959 -68 897 -132 4351 -66 5569 -66 2737 -66 1247 -66 423 -98 3809 -66 793 -66 4187 -66 4697 -66 457 -64 1477 -132 4585 -66 3481 -100 1525 -66 2385 -68 31825 -100 2139 -68 11545 -68 2955 -100 4449 -66 4171 -66 2339 -100 1921 -100 3617 -98 3277 -134 3089 -66 1095 -68 761 -100 297 -66 12083 -100 2161 -68 2459 -266 331 -100 5607 -100 1691 -98 1459 -100 331 -66 601 -130 1481 -66 331 -132 1515 -100 1221 -100 457 -66 531 -98 2209 -66 5969 -98 3395 -100 3191 -66 1619 -66 10377 -68 4111 -5242 261 -166 905 -394 97 -100 129 -228 131 -13052 265 -66 265 -134 97 -68 361 -132 229 -98 197 -132 1021 -98 1213 -66 2515 -68 2671 -98 3195 -98 919 -98 1315 -66 1083 -66 1415 -68 11665 -66 25317 -100 1121 -66 4129 -100 527 -166 1527 -130 2889 -66 893 -68
RAW_Data: 1717 -66 58363 -100 2423 -66 1987 -100 2313 -100 1615 -134 399 -66 6133 -66 2009 -66 2419 -66 3581 -132 495 -134 433 -132 9627 -100 997 -98 24859 -66 5677 -66 14265 -98 1859 -66 4807 -64 10307 -100 655 -98 6497 -132 99 -68 1621 -100 3485 -66 12641 -66 31045 -66 4923 -66 559 -98 6077 -68 1531 -66 14427 -66 15645 -66 2597 -132 1281 -66 1525 -68 429 -98 2751 -66 11147 -66 20825 -100 45999 -66 21363 -66 43717 -68 30743

```

Skorpionm, 2023-03-06:
The GateTx protocol must write 24 bits of data. You have 32, this error occurs on this error.

agilly, 2023-03-07:
I see. Does this mean the the protocol was not correctly identified, or that the sequence of bits was not correctly parsed? I see that the sequence has 4 leading null octets. Should I remove two and give it a try?

Skorpionm, 2023-03-09:
no, nothing is needed, it just says that your protocol is very very similar to gatetx , but in fact it is not

agilly, 2023-03-09:
I guess my question is, there is no way to store this information in
decoded way for this particular protocol? If I want to send it, I have to
record RAW?
The reason why I was interested in storing it decoded is that the decoded
format is easy to read and potentially edit, if I wanted to understand how
the protocol works i.e. build a dictionary codes to functions, etc.

Skorpionm, 2023-03-10:
the problem is that, since this is actually a GateTx protocol, information about the duration of encoding bits and the intervals between bursts is lost and the replay is different from the original

skotopes, 2023-03-13:
@agilly summarizing the issue, there are 3 different paths:

- Record and use Bin_Raw (new feature arrived couple releases ago, allows to extract meaningful part of the transmission. You can enable it in "Read" configuration)
- Record and use Raw 
- Add new or extend existing protocol (to do that we need raw/sdr recordings)