flipperdevices/flipperzero-firmware #2461: Allow in-place allocation of `FuriString`

dcoles, 2023-03-05:
### Describe the enhancement you're suggesting.

Currently the only way to create a `FuriString` is via a function like [`furi_string_alloc`](https://sourcegraph.com/github.com/flipperdevices/flipperzero-firmware@5d4057f7228b528896578ab6cf9d9a61b80a7c58/-/blob/furi/core/string.h?L35) that returns an opaque `FuriString*`. However the underlying [m-string](https://github.com/P-p-H-d/mlib#m-string) does support in-place allocation of `string_t`:

```c
{
    string_t s1;
    string_init (s1); // initialize `string_t` structure
    string_set_str (s1, "Hello, world!");
    string_clear(s1); // frees any memory dynamically allocated by `s1`
}
```

In-place allocation of `string_t` allows taking advantage of the short-string optimisation that allocates small strings directly in `string_t`, rather than always allocating space on the heap.

I believe it would be possible to add a `furi_string_init` and `furi_string_clear` function to the SDK that can initialize a stack-allocated `FuriString`. However this means that a stack-allocated `FuriString` can't be passed into any function that assumes it can take ownership of a `FuriString*` (and thus call `furi_string_free`). It would also require exposing the [implementation of `FuriString`](https://sourcegraph.com/github.com/flipperdevices/flipperzero-firmware@5d4057f7228b528896578ab6cf9d9a61b80a7c58/-/blob/furi/core/string.c?L4-6) so that the compiler can allocate the correct amount of space.

### Anything else?

The reason I'm interested in doing this is that it allows creation of a stack-allocated Rust `FuriString` type in [flipperzero-rs](https://github.com/flipperzero-rs/flipperzero) (see [discussion here](https://github.com/flipperzero-rs/flipperzero/pull/50#discussion_r1125574244)). The type would look approximately like the following (**note:** may have some syntax errors):

```rust
/// Dynamically allocated string.
/// This is a wrapper around the SDK's C `FuriString` type).
#[repr(transparent)]
pub struct FuriString(sys::FuriString);

impl FuriString {
    /// Create a new [`FuriString`]
    pub fn new() -> FuriString {
        // Uninitialized structure
        let mut furi_string = MaybeUninit<sys::FuriString> = MaybeUninit::uninit();
        unsafe {
            // Initialize `sys::FuriString` memory
            sys::furi_string_init(furi_string.as_mut_ptr());
        }
        
        // Wrap the initialized `sys::FuriString` in a wrapper type
        FuriString(unsafe { furi_string.assume_init() })
    }

    /// Get a pointer to this string that can be used with Flipper Zero SDK APIs.
    pub fn to_mut_raw(&mut self) -> *mut sys::FuriString {
        &mut self.0 as &FuriString as *mut sys::FuriString
    }
}

impl Drop for FuriString {
    fn drop(&mut self) {
        unsafe {
            // Ensure any dynamic memory allocations are freed
            sys::furi_string_clear(self.to_mut_raw());
        }
    }
}
```

skotopes, 2023-03-13:
In theory possible, but not going to save much.

DrZlo13, 2023-06-22:
1. SSO requires disclosed FuriString structure, but we want as much encapsulation as possible.
2. If we do not reveal the inners of the FuriString, but use the pointer as is - SSO will hold a maximum of 3-character strings.
3. When we migrated from M*Lib to our wrapper, I was profiling and the performance impact was very negligible.

Given all of the above - I see no point in this optimization.

closed 2023-06-22 by DrZlo13