flipperdevices/flipperzero-firmware #2465: Detect SL (sweden transport card) NFC

xSean145, 2023-03-05:
### Describe the enhancement you're suggesting.

Flipper Zero does not detect or read this card. Add support to detect and read this card.
Proxmark pulled this info from the card. (UID censored)

[usb] pm3 --> hf search
 🕕  Searching for ISO14443-A tag...          
[+]  UID: XX XX XX XX XX XX XX 
[+] ATQA: 00 44
[+]  SAK: 60 [1]
[+] MANUFACTURER: Infineon Technologies AG Germany
[=] -------------------------- ATS --------------------------
[+] ATS: 0B 78 80 74 03 80 54 05 30 3E 0F [ CF 00 ]
[=]      0B...............  TL    length is 11 bytes
[=]         78............  T0    TA1 is present, TB1 is present, TC1 is present, FSCI is 8 (FSC = 256)
[=]            80.........  TA1   different divisors are NOT supported, DR: [], DS: []
[=]               74......  TB1   SFGI = 4 (SFGT = 65536/fc), FWI = 7 (FWT = 524288/fc)
[=]                  03...  TC1   NAD is supported, CID is supported

[=] -------------------- Historical bytes --------------------
[+]   805405303E0F  (compact TLV data object)
[=]     54  05303E0F   Card issuer data

[?] Hint: try `hf mfu info`

[+] Valid ISO 14443-A tag found

[=] Short AID search:
[?] Hint: try emv commands

### Anything else?

It is recognized as a iso14443-4a.

Astrrra, 2023-03-07:
Does it not react to the card at all? Or does it at least show the UID?

xSean145, 2023-03-07:
It does not react at all to the card. I was told in Discord it might be something with the unusual SAK.
Can send you all the files pulled from the card there.

skotopes, 2023-03-13:
Yep, we do need more details. And most likely physical card.

xSean145, 2023-03-13:
I can send the physical card to anywhere in the world. Can I contact someone on Discord?

skotopes, 2023-03-13:
@xSean145 you can ping me skotopes#4130

bidhata, 2023-06-15:
any updates on this ?

skotopes, 2023-06-16:
@bidhata not really, we still don't have card for analysis.

idoo, 2023-10-03:
Has the issue been resolved with the latest release? 
https://github.com/flipperdevices/flipperzero-firmware/pull/3017

mxcdoam, 2024-01-31:
I have 4 units of SL cards (older MFC 1K and newer MF PLUS 4K)and I can read both on 0.97.1 FW (keys are available in system dict). No parser though.

skotopes, 2024-07-07:
@xSean145 do you still have the card? Does latest firmware works for you?

xSean145, 2024-07-07:
Still have the card. Never got a second adress to send it to. The first one i sent got returned to me because it was unclaimed. Will try new firmware once im back from traveling.

skotopes, 2024-07-07:
@xSean145 ok. Since we moved NFC team to our UK office you can send it here directly. Ping me when if you'll need address.