flipperdevices/flipperzero-firmware #2472: Inverted < > and \ | on Mac Italian keyboard

MrVideo, 2023-03-08:
### Describe the bug.

The keys `< >` and `\ |` are inverted on the `it-IT` keyboard layout on Mac. No USB Ducky script that uses those characters can work properly unless this issue is fixed.

For example, the `demo_macos.txt` script that comes with the device does not work properly. You can see the output of that script here:

```
❯ cat | /dev/null \\ EOF
zsh: permission denied: /dev/null
Hello World!
❯ ============================================================
zsh: =========================================================== not found
❯ 
❯               _.-------.._                    -,
zsh: command not found: _.-------.._
❯           .-"'''"--..,,_/ /'-,               -,  <
quote>        .:"          /:/  /'<  <     ,_...,  '. >  >
zsh: parse error near `<'
❯       /       ,----/:/  /'< _<'_-"'     _;
dquote>      '      / /'"""'< < <.'_-'      ,-"'/
zsh: permission denied: /
❯     >      > >  0    > > .-'      ,/'  /
zsh: parse error near `>'
❯    >    ,..< <     ,.-"'       ,/'    /
zsh: parse error near `<'
❯   ;    :    '/'""<'           ,/--==,/-----,
quote>   >    '-...>        -.___-Z:_______J...---;
zsh: no such file or directory:            ,/--==,/-----,\n  >    -...
❯   :         '                           _-'
❯  _L_  _     ___  ___  ___  ___  ____--"'
dquote> > __>> >   >_ _>> _ <> _ <> __>> _ <
dquote> > _> > >__  > > >  _/>  _/> _> >   /
dquote> >_>  >____>>___>>_>  >_>  >___>>_>_<
dquote> 
dquote> Flipper Zero BadUSB feature is compatible with USB Rubber Ducky script format
dquote> More information about script syntax can be found here:
dquote> https://github.com/flipperdevices/flipperzero-firmware/blob/dev/documentation/file_formats/BadUsbScriptFormat.md
dquote> EOF
dquote> 
```

By manually inverting these keys properly in the file, the script produces the correct output.

Would it be possible to get an Italian Mac keyboard layout in the future?

Thanks.

### Reproduction

1. Go to the main menu
2. Open the `Bad USB` application
3. Select `demo_macos.txt`
4. Run the script
5. The result should be similar to the one described above

optimus10ro, 2023-03-12:
Use ALTSTRING instead of STRING

ALTSTRING codes are language independent.

skotopes, 2023-03-13:
@nminaylov can you check keyboard map?

skotopes, 2023-06-22:
What we missing is it-IT-mac, it will be nice if someone contribute correct one.

nminaylov, 2024-01-29:
Please check this layout file https://github.com/flipperdevices/flipperzero-firmware/blob/ca05abea327cd34902a61c0294792a82b3bace22/applications/main/bad_usb/resources/badusb/assets/layouts/it-IT-mac.kl

skotopes, 2024-04-14:
Was fixed in https://github.com/flipperdevices/flipperzero-firmware/pull/3401

closed 2024-04-14 by skotopes