flipperdevices/flipperzero-firmware #2473: Custom message in furi_assert

polarikus, 2023-03-08:
### Description of the feature you're suggesting.

Sometimes a custom message is missing infuri_assert(), It will be easier to catch where something went wrong in the application if you write custom messages when the assertion failed.

Is it possible to add something like this?
#define furi_assertf(__e, message)                          \
    do {                                          \
        if(!(__e)) {                              \
            furi_crash(message); \
        }                                         \
    } while(0)
#else

hedger, 2023-03-10:
FYI: furi_assert/furi_crash doesn't preserve the message if it was called from a .fap -- since we are only storing a pointer to the message string, it'll be invalid after reboot.

hedger, 2023-03-10:
For better debugging experience, try using an SWD debugger  - WiFi board, any ST-Link, or even a second Flipper.

skotopes, 2023-03-13:
We'll think about it.

skotopes, 2023-06-30:
fixed in  #2819

closed 2023-06-30 by skotopes