flipperdevices/flipperzero-firmware #2484: nfc: MF Classic reading key B via A is not a valid operation

RouNNdeL, 2023-03-13:
### Describe the bug.

#2437 introduced a way to read the B key using the A key. This is possible but not useful in a key attack. The specification explicitly mentions that when  the B key is readable by the A key it cannot be used for authentication and can be used as additional 6 data bytes. 

### Reproduction

1. Read #2437

### Anything else?

From the specification:

> If key B may be read in the corresponding Sector Trailer it cannot serve for authentication (see grey marked lines in Table
7). As a consequences, if the reader authenticates any block of a sector which uses such access conditions for the Sector
Trailer and using key B, the card will refuse any subsequent memory access after authentication.

AloneLiberty, 2023-03-18:
If key B is readable, it can't be used to authenticate **ONLY** on the sector where it can be read. But this key may be reused on other sectors.
You should read more carefully:
If the reader authenticates any block of a sector **which uses such access conditions** for the sector trailer the card will refuse any subsequent memory access after authentication.

You can also check my example from verification:
>Write sector with access bits that allow reading key B, example:
>Block 3: A0A1A2A3A4A5FF0780001122FF112299
>Block 7: 1122FF112299778F0800FF11223344AA
>Try reading tag with Flipper, it should read 3 keys and 2 sectors

RouNNdeL, 2023-03-19:
Makes sense, thanks for explaining it.

closed 2023-03-19 by RouNNdeL