flipperdevices/flipperzero-firmware #2487: Incorrect comparison in subghz_setting_get_hopper_frequency

xarrange, 2023-03-14:
### Describe the bug.

Current implementation is:

```
uint32_t subghz_setting_get_hopper_frequency(SubGhzSetting* instance, size_t idx) {
    furi_assert(instance);
    if(idx < FrequencyList_size(instance->frequencies)) {
        return *FrequencyList_get(instance->hopper_frequencies, idx);
    } else {
        return 0;
    }
}
```

while it should be

```
uint32_t subghz_setting_get_hopper_frequency(SubGhzSetting* instance, size_t idx) {
    furi_assert(instance);
    if(idx < FrequencyList_size(instance->hopper_frequencies)) {
        return *FrequencyList_get(instance->hopper_frequencies, idx);
    } else {
        return 0;
    }
}
```

idx should be comapared to hopper frquencies list and not to main frequencies list

### Reproduction

1. Change hopper list so that it would longre than main frequency list
2. Run app with hopping

skotopes, 2023-03-16:
Nice catch

Skorpionm, 2023-03-20:
fix https://github.com/flipperdevices/flipperzero-firmware/pull/2518

closed 2023-03-20 by skotopes