flipperdevices/flipperzero-firmware #2501: Increased power consumption after disconnecting GPIO module

123jjck, 2023-03-17:
### Describe the bug.

My Flipper's default power consumption is 9 mA. 
After connecting and disconnecting ESP32-WROOM to GPIO (3.3V, GND, TX and RX), the consumption becomes 2-3 mA more, i.e. 11-12 mA and does not drop until the Flipper reboot.
5V on GPIO is off

### Reproduction

1. Connect ESP32-WROOM to Flipper (i use [that ](https://github.com/UberGuidoZ/Flipper/blob/main/GPIO/ReadMe.md#esp32--esp32-wroom-pinout-to-flipper-zero)pinout)
2. Disconnect
3. Check power consumption

skotopes, 2023-03-17:
Most likely this caused by SD-Card being reseted by power glitch on connect/disconnect. Try without SD Card.

123jjck, 2023-03-17:
> Try without SD Card

Doesn't help, same problem.

skotopes, 2023-03-20:
Caused by USART subsystem waiting for someone to get data from RX pipe.

PR #2523 (2023-03-22): [#2501] Disable UART IRQs by default
# What's new

- UART IRQ events are disabled by default (should fix #2501)

# Verification 

- Open settings -> power info, try sending some data to UART console
- Consumption power should remain at 9-10ma
- Check USB-UART bridge app

# Checklist (For Reviewer)

- [x] PR has description of feature/bug or link to Confluence/Jira task
- [x] Description contains actions to verify feature/bugfix
- [x] I've built this code, uploaded it to the device and verified feature/bugfix