flipperdevices/flipperzero-firmware #2502: Launching external FAPs from the serial CLI

str4d, 2023-03-17:
The `loader` CLI can be used to run applications that are built into the firmware. However, it does not appear to be possible to run applications built as external FAPs. When such an application is running, the CLI reports:
```
> loader info
Running application: Applications
```
and opening the FAP directly doesn't work:
```
> loader open /path/to/the.fap
/path/to/the.fap doesn't exist
```

It would be useful to be able to run external FAPs via the serial CLI, and be able to provide them arguments. If this is already possible and I just haven't found out how, it would be useful to update the documentation to explain how to achieve this.

DrZlo13, 2023-03-17:
You can use `loader open Applications /path/to/the.fap`

closed 2023-03-17 by hedger

str4d, 2023-03-17:
> Passing arguments to .fap files is not supported at the moment. You can create a separate issue on that.

Done: #2505.