flipperdevices/flipperzero-firmware #2505: Support passing arguments via serial CLI to external FAPs

str4d, 2023-03-17:
All applications that aren't run on startup are required to have an entry point of the form `int32_t entry_point(void* p)`:
https://github.com/flipperdevices/flipperzero-firmware/blob/ccaa3864d54b3792b8ab0ec0a2efbd82b3c2f7ae/scripts/fbt/appmanifest.py#L354-L357

This implies that some kind of context or arguments object is provided by the Flipper Zero to the app via the `void* p`. This appears to be the case for built-in apps; for example, the iButton app checks if `p` is a C string, and then parses it:
https://github.com/flipperdevices/flipperzero-firmware/blob/224d0aefe41edd8356aa41aa6b39cbe7659c5d98/applications/main/ibutton/ibutton.c#L313-L336

However, when running an external FAP from the serial CLI, providing arguments causes it to fail to start:
```
> loader open Applications /path/to/the.fap some args
// On screen
Error
Preload failed
Invalid file
```

Allowing arguments to be passed when starting via the serial CLI would enable external FAPs to act similarly to built-in CLI apps (albeit with the `loader open Applications` prefix).

Alternatively, if there were a way for an external FAP to register a CLI interface (like is often done in apps with type `FlipperAppType.STARTUP`), that would satisfy the use case. However, I suspect this would be trickier as it would require scanning the entire directory tree to find external FAPs to register in this way, so passing arguments directly to the external FAP is likely to be simpler.

hedger, 2023-03-17:
Yep, we're also planning on implementing fixes to add RPC communication to .faps.

str4d, 2023-08-11:
AFAICT this is supported after the FAP loader refactor in 0.86.1: because the loader is now doing the FAP loading, its argument handling Just Works. Thanks!

closed 2023-08-11 by str4d