flipperdevices/flipperzero-firmware #2510: Upside down screen in qflipper and Flipper Lab when using lefty mode

Milk-Cool, 2023-03-18:
### Describe the bug.

When using lefty mode, the screen display in qflipper is upside down.

[image]

### Reproduction

1. Eanble lefty mode in settings
2. Connect you flipper to a comouter
3. Install and open qflipper

### Target

Flipper Zero FZ.1 firmware version 0.79.1 with lefty mode enabled

skotopes, 2023-06-06:
Should be fixed in all companion apps

closed 2023-06-06 by skotopes

pyrophreak, 2023-06-24:
There is a Feature in Xtreme Firmware if your screen is flipped in the settings  menu there are 2 arrows in a circle that give you the option to flip it back and forth.