flipperdevices/flipperzero-firmware #2513: U2F over Bluetooth

Kami-no, 2023-03-19:
### Description of the feature you're suggesting.

It would be great to have BT support because wires are boring and MacBook has no NFC.

evgfilim1, 2023-10-07:
I'll try to implement it. @skotopes can you assign it to me?

If you don't hear from me anything during about a month, it means I'm too busy and someone can take the task.

skotopes, 2023-10-08:
We'll wait for your PR ;-)

xnohat, 2024-01-04:
Can we implement this feature @skotopes  ? U2F over bluetooth a great because on market only a fews option.

skotopes, 2024-01-05:
@xnohat we can implement anything, but that doesn't mean that we should.

In this specific case there are some concerns:
- U2F is superseded by FIDO. That makes implementing FIDO higher priority task.
- Bluetooth in our specific case means BLE, before starting U2F/FIDO over BLE I'd like to know how many systems are on the market and how well whole thing is widespread

As far as I understand whole thing is a very niche(same as U2F over NFC), which makes is low priority.

huan, 2024-01-22:
I'm so looking forward to using U2F over Bluetooth on my Mac! (I just got my Flipper this week, and it's so good that far more than my expectation!)

WestonGFX, 2024-05-10:
This is something I would like to do U2F over Bluetooth on MacOS as well with my F0.

lzlrd, 2024-08-23:
https://github.com/fmeum/WearAuthn might be a good reference, supports FIDO[2] over Bluetooth and NFC (https://github.com/flipperdevices/flipperzero-firmware/issues/1565).

rouing, 2024-12-26:
I use U2F over BLE and NFC regularly. More so NFC as it's our way of doing authentication into AD Domains. Just putting my 2c