flipperdevices/flipperzero-firmware #2515: Broken "OK" button on "Mounting SD card failed" screen

ValdikSS, 2023-03-19:
### Describe the bug.

"OK →" button on "Mounting SD card failed" screen does not do anything. The screen does not disappear when pressing → D-pad on said screen, while all other buttons work fine.

### Reproduction

1. Insert completely blank MicroSD card (no partition layout) while Flipper is powered off
2. Power Flipper on
3. See "Mouting SD card failed" screen with "OK →" button
4. Press → D-pad button
5. Nothing happens. The screen does not disappear.

### Target

0.79.1

Banditwon, 2023-09-12:
Did you figure out the issue?

ValdikSS, 2023-09-12:
I've partitioned the card and created the file system on a PC. I didn't fix the bug.

Banditwon, 2023-09-12:
Okay, thanks I will try that out.I was using the GPIO then all off a sudden it stopped working. I received that error message, now I can't update the flipper to a different firmware because I can't mount the sd card.

tudorsss, 2023-12-09:
Happens for me too

chrisbavaria, 2023-12-13:
While updating my Flipper, I have same problem. No SD-Card is working, also the working SD-Card before updating and fresh formating with FAT32.
Settings  > Storage > Mount SD Card > Unmount: The LED blinks a short time Green,
Settings > Storage > Mount SD Card > Mount: The LED The LED blinks 10x a long time purple and blinking 3x short red.
Then the Screen shows "Cannot Mount SD Card" "internal error"
Is there a solution for this problem?

Banditwon, 2023-12-13:
I gave up on customer support and just got another one.

skotopes, 2024-07-07:
> I gave up on customer support and just got another one.

can you send me your ticket number?

For anyone who will have any issues with SD card please start here: https://docs.flipper.net/basics/sd-card

ValdikSS, 2024-07-07:
@skotopes there aren't issues with the card, the bug is in firmware code.

skotopes, 2024-07-07:
You can not skip Mounting SD Failed animation if such SD card is inserted. 
From the code perspective it's more a feature than a bug: system can not function properly if card failed to mount.
We'll remove OK button in next release.

skotopes, 2024-07-09:
Fixed in  #3776

closed 2024-07-09 by skotopes