flipperdevices/flipperzero-firmware #2525: NFC Writing Function Status

berkut0, 2023-03-22:
### Description of the feature you're suggesting.

Hi

I would like to know the status of the NFC recording capability. I have searched a bit among the issues, but have not found a clear answer. On the Internet, too, write different things, for example, that this feature is "about to" be implemented.

1. This feature has not been implemented?

2. Will this feature be implemented one day? If so, should we expect it anytime soon?

3. Are there alternate firmware versions that support this feature?

4. Are you not implementing this feature to avoid legal or competitor issues?

skotopes, 2023-03-22:
What type of the card are we talking about?

berkut0, 2023-03-23:
@skotopes sorry, my bad, I think I haven't paid enough attention to it
I'm talkinkg about mostly Mifare Classic 1K, but also interested in another cards

Now I see the "write to initial card" option at Mifare Classic 1K, but the question then is whether I can write along with the UID to an empty card (yes, I have a card with rewritable UIDs)?

Also, it seems that Mifare Ultralight does not have an option to write. As far as I know, it is not a card limitation.

berkut0, 2023-03-24:
I have done some research and now I understand what the problem was.
To work with Mifare Ultralight, I have to **scan the reader** and then unlock with the key calculated with the **mfkey32** tool?

skotopes, 2023-03-28:
yep, Nfc Magic is currently supporting only 1st gen of magic cards, what you want sounds like 4th generation (ultimate magic card). There is a work in progress PR with gen4 support.

closed 2023-03-28 by skotopes

berkut0, 2023-04-08:
@skotopes 
I apologize for breaking into a closed topic. Will gen2 be supported or does gen4 automatically support gen2?