flipperdevices/flipperzero-firmware #2531: Regression bug in mifare NFC

alexandergre, 2023-03-24:
### Describe the bug.

The RC 0.80.0 and later versions have a bug which is not present in release version 0.79.1. With release version, flipper can find all keys and read all sectors from mifare card in under 10 seconds. On RC however this process takes a long time and flipper fails to find all the keys and unlock all sectors.

For testing, I'm using the same card, and I never save the card so flipper doesn't cache the keys.

I checked the .keys file in nfc cache folder (I deleted the file afterwards), key B is FF FF FF FF FF FF and key A is A0 A1 A2 A3 A4 A5 for sector 0 and then changes to D3 F7 D3 F7 D3 F7 for the rest of the sectors except 15, where it is FF FF FF FF FF FF. Here is this file:

`Filetype: Flipper NFC keys
Version: 1
Mifare Classic type: 1K
Key A map: 000000000000FFFF
Key B map: 000000000000FFFF
Key A sector 0: A0 A1 A2 A3 A4 A5
Key B sector 0: FF FF FF FF FF FF
Key A sector 1: D3 F7 D3 F7 D3 F7
Key B sector 1: FF FF FF FF FF FF
Key A sector 2: D3 F7 D3 F7 D3 F7
Key B sector 2: FF FF FF FF FF FF
Key A sector 3: D3 F7 D3 F7 D3 F7
Key B sector 3: FF FF FF FF FF FF
Key A sector 4: D3 F7 D3 F7 D3 F7
Key B sector 4: FF FF FF FF FF FF
Key A sector 5: D3 F7 D3 F7 D3 F7
Key B sector 5: FF FF FF FF FF FF
Key A sector 6: D3 F7 D3 F7 D3 F7
Key B sector 6: FF FF FF FF FF FF
Key A sector 7: D3 F7 D3 F7 D3 F7
Key B sector 7: FF FF FF FF FF FF
Key A sector 8: D3 F7 D3 F7 D3 F7
Key B sector 8: FF FF FF FF FF FF
Key A sector 9: D3 F7 D3 F7 D3 F7
Key B sector 9: FF FF FF FF FF FF
Key A sector 10: D3 F7 D3 F7 D3 F7
Key B sector 10: FF FF FF FF FF FF
Key A sector 11: D3 F7 D3 F7 D3 F7
Key B sector 11: FF FF FF FF FF FF
Key A sector 12: D3 F7 D3 F7 D3 F7
Key B sector 12: FF FF FF FF FF FF
Key A sector 13: D3 F7 D3 F7 D3 F7
Key B sector 13: FF FF FF FF FF FF
Key A sector 14: D3 F7 D3 F7 D3 F7
Key B sector 14: FF FF FF FF FF FF
Key A sector 15: FF FF FF FF FF FF
Key B sector 15: FF FF FF FF FF FF
`

To demonstrate, this is version 0.79.1: https://i.imgur.com/A10AUgu.mp4

This is how 0.80.0 and up behaves (I shot two videos to show start and end of this process because it’s too long): https://i.imgur.com/iFWeUuG.mp4 https://i.imgur.com/8yPxGKn.mp4

### Reproduction

Use NFC read option on an NFC card

Astrrra, 2023-03-24:
Hi! I haven't been able to reproduce this issue with the info you've provided. Can you please attach a full card dump from a release firmware version here?

alexandergre, 2023-03-26:
I have tested the build from the fix branch, it works!

PR #2533 (2023-03-27): [FL-3193] Additional checks before invalidating the key
# What's new

- Fixes #2531 

# Verification 

- Read the card from the thread, check that it gets read properly
- Read other cards, check that nothing broke

# Checklist (For Reviewer)

- [x] PR has description of feature/bug or link to Confluence/Jira task
- [x] Description contains actions to verify feature/bugfix
- [x] I've built this code, uploaded it to the device and verified feature/bugfix
closed 2023-03-27 by skotopes