flipperdevices/flipperzero-firmware #2535: LCD Backlight off when using phone control

xhd4, 2023-03-24:
### Description of the feature you're suggesting.

Turn off the backlight of the display when controlled from the phone. Is it possible to implement a function to turn off the backlight of the flipper screen when using phone control? But at the same time, the backlight will turn on after using the keys of the flipper itself. And add this setting as an item in Settings -> LCD and Notifications

skotopes, 2023-03-28:
And use cases are?

xhd4, 2023-03-28:
accidentally closed

xhd4, 2023-03-28:
accidentally closed

skotopes, 2023-06-22:
After some thought this feature looks quite niche and complex in implementation to make it right. So I'll close this issue.

closed 2023-06-22 by skotopes