flipperdevices/flipperzero-firmware #2538: Broken "OK" button when defining a pin with GpioModeInterruptRise mode

EstebanFuentealba, 2023-03-26:
### Describe the bug.

When defining pin C3 in **GpioModeInterruptRise** mode and exiting the application, the OK button no longer works, requiring a reboot of the Flipper Zero.

### Reproduction

I added a repository with a basic application that reproduces the error: https://github.com/EstebanFuentealba/Flipper-Zero-Bug-GpioModeInterruptRise

Steps to reproduce:

1. Clone & Run repository
2. Open the app.
3. Close the app.
4. The `OK` button becomes unresponsive.

### Target

0.79.1

EstebanFuentealba, 2023-03-26:
https://github.com/flipperdevices/flipperzero-firmware/issues/2538

skotopes, 2023-03-28:
That is expected behavior. PC3 uses EXTI3 line which is shared among all Px3 pins on all ports.
Also pay attention to https://docs.flipperzero.one/gpio-and-modules all shared EXTI lines are marked there.

We'll add additional checks to prevent pins from incorrect configuration.

closed 2023-03-28 by skotopes