flipperdevices/flipperzero-firmware #2542: Mifare Classic Write Support

optimus10ro, 2023-03-28:
### Description of the feature you're suggesting.

I want to be able to read a tag and write it on other NFC tags, I could copy my gate key, and other stuff.

AloneLiberty, 2023-03-29:
If you want to copy your tag to new tag, you will need tag that can change UID - Magic Gen1 tag. https://docs.flipperzero.one/nfc/magic-cards

optimus10ro, 2023-03-29:
I want to be able to write saved cards

AloneLiberty, 2023-03-29:
If you want to update data on your card from data that saved on Flipper you can use "Write to initial card" 
If you want to write saved data to new card you must use gen1 with Nfc magic app (other types of magic cards aren't supported yet). There is no way to change UID on regular MFC cards

AloneLiberty, 2023-03-29:
If you want to write to another card you must use gen1 magic tag. Read docs for more details https://docs.flipperzero.one/nfc/magic-cards

skotopes, 2023-03-29:
> O want to write the data to another card, like create another key that has the same data

Normal cards doesn't allow to write block 0.

If it's ok, then dump new card, then copy blocks from old dump to new, then write new dump to new card

closed 2023-03-29 by skotopes