flipperdevices/flipperzero-firmware #2545: SPI Mem Manager Write Protection

Roemrat, 2023-03-28:
### Describe the bug.

Can't write or erase with write protection IC.

### Reproduction

write or erase with write protection IC.

Ethanol0001, 2023-03-28:
we need more details in the report, just saying write or erase doesn't help. Add a more detailed description along with a step by step guide on how to reproduce it, along with adding the logs for the bug too.

dogtopus, 2023-03-29:
Did you pull the write protect pin up?

Roemrat, 2023-03-29:
When you set "SRWD bit (Status Register Write Disable)" e.g. 0x9C, 0x8C, 0xFC, you can't write or erase.
normally it must be set 0x00 (Unprotect) before write or erase.
[image]

closed 2023-04-01 by skotopes