flipperdevices/flipperzero-firmware #2546: Record entire IR signal for later analysis and replay

C0ntr07, 2023-03-29:
### Description of the feature you're suggesting.

I tried to record and reply my TV-B-Gone.  Unfortunately, Flipper kept interpreting the IR codes and not recording the entire (very long) transmission.

An extended feature to record an entire set of collected IR signals, safe them, and then later replay them would be good.  Maybe also have a way to take the raw data and provide annotations where possible would also be beneficial.

skotopes, 2023-03-29:
In theory it's possible, furi_hal_infrared provides such option. We'll try to implement.

gsurkov, 2023-03-30:
@C0ntr07 for now, you can use the `ir` command from Flipper CLI. It has a special form `ir rx raw` which allows receiving any signals without decoding them.

Saijin-Naib, 2023-07-11:
Would this be implemented as a Flipper app?

I'm looking to replace my very worn DangerousPrototypes USB Infrared Toy, so being able to capture/log Infrared and then play them back selectively would be sublime.

skotopes, 2023-07-11:
@Saijin-Naib we don't have plans to implement it by our self. All bits and pieces are already here, maybe someone from community will be willing to implement it.

Meanwhile I'm closing this issue.

closed 2023-07-11 by skotopes